zeroflag/punyforth #2: cant follow your step on linux

yunfan, 2016-08-19:
will you please add linux support?

zeroflag, 2016-08-19:
Could you elaborate further, what kind of linux support do you mean? Currently you can run punyforth on linux, but there is no os integration (socket, file system, etc).

# cd punyforth/arch/x86
# ./build

Thils will compile, test, and start the REPL on linux.

yunfan, 2016-08-20:
no i mean build that firmware on linux and flashing them to my module using esptool

also when i compiled it on my cellphone(termux on android) it fails too, and mostly caused by assembly syntax

zeroflag, 2016-08-20:
> no i mean build that firmware on linux and flashing them to my module using esptool

Ok, there are no build instructions yet. Bu basically you'll need to clone and build this project 

https://github.com/SuperHouse/esp-open-rtos

This needs esp-open-sdk, just follow the instructions on the github page. After that copy punyforth into the example directory of esp-open-rtos and run make.

There will be detaild instructions about the build process at some time.

yunfan, 2016-08-21:
so your forth is an rtos app not barebone one?

zeroflag, 2016-08-21:
Yes, this is a programming language, not an operating system.

yunfan, 2016-08-21:
i followed all the ins of esp-open-rtos

and then i got this after i enter the `examples/punyforth` then type `make`

still got error 

```
jyf@brix:~/repo/git/esp-open-rtos/examples/punyforth$ LC_ALL=c  make
make: *** No targets specified and no makefile found.  Stop.
```

zeroflag, 2016-08-21:
The makefile is located at esp-open-rtos/examples/punyforth/arch/esp8266/rtos/user

Try to run make in this directory.

yunfan, 2016-08-22:
yes i have enter that directory and type make too
still lack of a file named uber.forth

zeroflag, 2016-08-22:
> yes i have enter that directory and type make too
> still lack of a file named uber.forth

The uber.forth is generated by the modules.py python script. It is just a big (zero terminated) source file that contains all of the selected modules in the correct order. Use the modules.py to generate one then you'll be able to flash it with help of the flash script (it is in the same directory as the modules.py).

zeroflag, 2016-10-31:
I added build instructions to the wiki

https://github.com/zeroflag/punyforth/wiki/Build-environment-setup

closed 2016-10-31 by zeroflag

yunfan, 2016-11-01:
`export PATH=$PATH:/esp/esp-open-sdk/xtensa-lx106-elf/bin`

should be 

`export PATH=$PATH:~/esp/esp-open-sdk/xtensa-lx106-elf/bin`

zeroflag, 2016-11-01:
@yunfan thx, fixed

mhoffma, 2017-03-26:
what is uber.forth it seems to be missing?  is that just any forth program from the examples directory?