zeroflag/punyforth #3: Want to understand, how it works

Spray27ds, 2016-11-29:
Hallo,
great that there is a forth for ESP8266 :-)
I want to use it for personal development and will try to add mutt-support.
Browsing through the source I wonder how/when uber.forth is read from flash during startup.
Can you give me some hints, so I can better understand how it woks?

mny tnx in advance,

Dietmar

zeroflag, 2016-11-29:
hi,

Sorry for my ignorance, what is mutt-support?

The uber.forth is written to the flash (at a predefined address 0x51000) by the flash.bat/flash.sh script.

Punyforth always reads the source using the "key" word. During the boot, the key word is redirected to read from the flash instead of uart.

Take a look at next_char_from_flash(), forth_getchar(), forth_getchar_nowait() in https://github.com/zeroflag/punyforth/blob/master/arch/esp8266/rtos/user/forth_io.c

The uber.forth is ended with a \0 character. When this character is read, the "key" word will read from the uart again.

Attila

Spray27ds, 2016-11-29:
Hi Attila,

thanks for your fast answer, now I understand :-)

I meant MQTT support, sorry for the typo.

All the best,

Dietmar

<author@noreply.github.com>
Datum:  29.11.2016 17:28
(#3)

https://github.com/zeroflag/punyforth/blob/master/arch/esp8266/rtos/user/forth_io.c

zeroflag, 2016-11-29:
cool, let me know if you have any further questions

closed 2016-12-11 by zeroflag