zeroflag/punyforth #4: traceback seems to have problems printing stack trace

Spray27ds, 2016-12-03:
Hello,

"traceback" seems to have problems to print the names of the word in the stack trace.

Example:
` exception: EXC
: test-trace EXC traceback ;
test-trace
`
prints:
> Exeption: exc rdepth: 3
>   at ??? (1075948848)
> Restarting ESP ..
> 

If I do 
`: testrp rp@ @ heap-end - . ;
testrp`
it shows that the address on top is about 2MB above the heap-end.
I do not understand whats going on here. The addresses on the return stack seem to be not inside the dictionary.

Can you explain, what I'm doing wrong here?
Many thanks,

Dietmar

zeroflag, 2016-12-04:
Hi,

This is not what traceback is meant for. Traceback is used internally to give a return stacktrace upon an unhandled exception. Although I think it should work that way too.

For example:

```forth
(stack) exception: EXC
(stack) : w1 EXC throw ;
(stack) : w2 w1 ;
(stack) : w3 w2 ;
(stack) : w4 w3 ;
(stack) : w5 w4 ;
(stack)
(stack) w5

Exeption: EXC rdepth: 9
  at unhandled (140296)
  at throw (140488)
  at w1 (148484)
  at w2 (148508)
  at w3 (148528)
  at w4 (148548)
```

w5 is not visible in the trace for some reason, this is probably a bug that need to be investigated.

If I add some nested calls to your example it works but has the same problem.

```forth
(stack) exception: EXC
(stack)
(stack) : w1 EXC traceback ;
(stack) : w2 w1 ;
(stack) : w3 w2 ;
(stack) : w4 w3 ;
(stack) : w5 w4 ;
(stack)
(stack) w5

Exeption: EXC rdepth: 7
  at w1 (148484)
  at w2 (148508)
  at w3 (148528)
  at w4 (148548)
```

The reason for not working this with only one call might be the same that causes that the last call is not visible in the nested case. I'll take a look at the problem more deeply sometime later.

zeroflag, 2016-12-04:
```forth
(stack) exception: EXC
(stack) : w1 EXC throw ;
(stack) : w2 w1 ;
(stack) : w3 w2 ;
(stack) : w4 w3 ;
(stack) : w5 w4 ;
(stack)
(stack) w5
```

This is what happens:

* We call w5 from the outer interpreter.
* w5 calls w5 but pushes the return address

zeroflag, 2016-12-11:
I modified the traceback word a bit. Now it gives the following output for this code

```forth
exception: E
: a 10 0 do i 5 = if E throw then loop ;
: b a ;
: c b ;
: d c ;
: e d ;
e
```
```
Exeption: E rdepth: 10
  at throw (1073648592)
  at a (1073653432)
  at ??? (5)
  at ??? (10)
  at b (1073653500)
  at c (1073653520)
  at d (1073653540)
  at e (1073653560)
```

This is the correct trace. As you can see now, the call ('e') is presented in the stack trace. (the number 5 and 10 are the loop variables).

Without nested calls:

```forth
exception: EXC : test-trace EXC throw ; test-trace
```
```
Exeption: EXC rdepth: 4
  at throw (1073648592)
  at test-trace (1073653388)
```

The usage of the traceback word might be revised in the future because currently it's not that useful from a user perspective.

closed 2016-12-11 by zeroflag