zeroflag/punyforth #8: unloop problem in nested do loops

CraigLindley, 2016-12-30:
Unless I'm missing something it seems like there is a bug in unloop when used in nested do loops. The code below illustrates the problem.

: test 20 0 do i . cr 10 0 do i . i 4 = if unloop exit then loop loop ;

zeroflag, 2016-12-30:
This is normal in classical Forth. 

https://forth-standard.org/standard/core/UNLOOP

"Discard the loop-control parameters for the current nesting level. **An UNLOOP is required for each nesting level** before the definition may be EXITed"

I'll update the readme because do loop must be understood internally otherwise it is very easy to mess things up (there are other constructs that don't work well with do loops).

CraigLindley, 2016-12-30:
I was trying to use unloop exit as the equivalent of leave which I had in
my original code. Is there some reason you didn't implement leave ?

-- 
Craig Lindley / Heather Hubbard

495's Recordings: craigandheather.net/495spage.html
New Recordings: craigandheather.net/cnmpage.html
Latest rock CD: craigandheather.net/oneinarow.html
Latest non-rock CD: craigandheather.net/craigdoesfingerstyle.html

Personal Website: craigandheather.net

Phone: (719) 495-1873
Cell: (719) 502-7925

If you’re one in a million, there are now seven thousand people exactly
like you.

zeroflag, 2016-12-30:
I haven't needed it yet. I like factoring out the logic in a new word and use exit to exit from the word instead of just the loop.

zeroflag, 2016-12-30:
I updated the readme with some notes on using unloop and exit from nested loops.

closed 2016-12-30 by zeroflag