zeroflag/punyforth #9: Enhancement request - add r@ word

CraigLindley, 2017-01-02:
I seem to use this word often

zeroflag, 2017-01-02:
r@ is like rp@ @ ?

CraigLindley, 2017-01-02:
Yes just grabbing the value on the top of the return stack without popping
it off. I coded it this way but I thought a primitive function would be
better.

-- 
Craig Lindley / Heather Hubbard

495's Recordings: craigandheather.net/495spage.html
New Recordings: craigandheather.net/cnmpage.html
Latest rock CD: craigandheather.net/oneinarow.html
Latest non-rock CD: craigandheather.net/craigdoesfingerstyle.html

Personal Website: craigandheather.net

Phone: (719) 495-1873
Cell: (719) 502-7925

If you’re one in a million, there are now seven thousand people exactly
like you.

zeroflag, 2017-01-02:
Ok. Do you need it for convenience or performance reasons? Currently we have about 35 essential primitives and I'd like to keep this number as low as possible because of portability reasons. An other reason I wouldn't implement this as a primitive because a future plan is to add some treeshaking or dead code elimination tool which would remove all the words from the uber.forth which are not used. And  a primitive cannot be removed by this technique because they are compiled into the binary. The heap size on the esp is 24k and it's quite a limiting factor, you can quickly fill it up by loading 3-4 modules and defining some new words.

Implenting this as a word is doable (there will be already a return address on the rstack and the real value is below that) but it would be slower.

CraigLindley, 2017-01-02:
It is for convenience as all of the other forths I have used have had the
set of return stack operators: >r, r@ r>. If it doesn't fit into your
architecture that's OK I have coded it like this and it works.

: r@ ( -- n )
    r> r> dup >r swap >r ;

-- 
Craig Lindley / Heather Hubbard

495's Recordings: craigandheather.net/495spage.html
New Recordings: craigandheather.net/cnmpage.html
Latest rock CD: craigandheather.net/oneinarow.html
Latest non-rock CD: craigandheather.net/craigdoesfingerstyle.html

Personal Website: craigandheather.net

Phone: (719) 495-1873
Cell: (719) 502-7925

If you’re one in a million, there are now seven thousand people exactly
like you.

zeroflag, 2017-01-29:
I added r@ to ext.S. This contains the nonessential primitives and some extra words. I realized this doesn't effects the size of the usable dictionary space.

closed 2017-01-29 by zeroflag