zeroflag/punyforth #10: Enhancement Request - Escape " in strings

CraigLindley, 2017-01-09:
The application I am writing uses a lot of strings. In this app I need to embed double quotes in other strings. I need to do things like: "\"Hello World\""

To this end I made the following change to core.frt

```
: eschr ( char -- char ) \ read next char from stdin
    dup [ char: \ ] literal = if
        drop key case
            [ char: r ] literal of 13 endof
            [ char: n ] literal of 10 endof
            [ char: t ] literal of 9  endof
            [ char: \ ] literal of 92 endof
            [ char: " ] literal of 34 endof  \ CAL
            EESCAPE throw
        endcase
    then ;
```

zeroflag, 2017-01-09:
An other option is just use single single quotes in that case ( str: 'asdf
"qwer" zxc'). But adding escape support makes sens too. I'll check if it
brakes something, and if not I'll integrate it.

zeroflag, 2017-01-13:
integrated

closed 2017-01-13 by zeroflag