zeroflag/punyforth #20: No interactive with punyforth

gogonkt, 2017-03-05:
Hey Dear!

I flashing and flashing, search and search tonight but still no idea what's wrong and need some help please.

I installed driver of CH340G for mac
$ python modules.py test
or
$ python modules.py core
and then 
$ ./flash.sh /dev/tty.wchusbserial1a20 #flashing my esp8266
and then 
$ screen /dev/tty.wchusbserial1a20 115200
or
$ picocom -b 115200 -f h /dev/tty.wchusbserial1a20 (for dupble check)

punyforth said:
$ screen /dev/tty.wchusbserial1a20 115200

test:core-str-escape
test:core-multi-line-str
test:eval-whitespace
test:core-str
test:core-alloc
test:core-rdepth
test:core-catch
test:core-defer
test:core-case
test:core-hex
test:core-factorial
test:core-between
test:core-logic
test:core-unloop
test:core-doloop
test:core-?dup
test:core-bounds
test:core-branch
test:core-arithmetic
32 tests, 32 passed, 0 failed, 0 errors
All passed

(stack) Punyforth ready.
(stack) 

but I can got any interactive with her :( 
nothing hapen when press any key. and sometime someone joined this AP XD
Orz

Thank you!

zeroflag, 2017-03-05:
I use the following settings in termite (this is a windows client) and coolterm (this is crossplatform): 115200 bps. Local echo: on, line mode: enabled.

I think screen uses raw mode instead of line mode, and I don't know if it can be reconfigured to use line mode instead. If you can find out, please let me know, because I'm interested too. Otherwise I recommend using coolterm or an other client that can be configured this way.

gogonkt, 2017-03-06:
Hey Dear!

I found that 115200 bps yes; 
                   Local echo: on, not why, it only show you what you are typing, "picocom -c ... " solved.
                   line mode: enabled. I think it why that punyforth no react and that is cr lf problom. what crlf is punyforth needed?

man pico:
INPUT, OUTPUT, AND ECHO MAPPING
       Using the --imap, --omap, and --emap options you can  make  picocom  map
       (tranlate, replace) certain special characters after being read from the
       serial port (with --imap), before being written to the serial port (with
       --omap),  and before being locally echoed to the terminal (standard out-
       put) if local echo is enabled  (with  --emap).   These  mapping  options
       take,  each, a single argument which is a comma-separated list of one or
       more of the following identifiers:

       o crlf (map CR to LF),

       o crcrlf (map CR to CR + LF),

       o igncr (ignore CR),

       o lfcr (map LF to CR),

       o lfcrlf (map LF to CR + LF),

       o ignlf (ignore LF),

       o bsdel (map BS to DEL),

       o delbs (map DEL to BS)

       For example the command:

              picocom --omap crlf,delbs --imap inglf,bsdel --emap crcrlf ...

       will:

       o Replace every CR (carriage return, 0x0d) caracter with LF (line  feed,
         0x0a) and every DEL (delete, 0x7f) character with BS (backspace, 0x08)
         before writing it to the serial port.

       o Ignore (not write to the terminal) every LF character  read  from  the
         serial  port, and replace every BS character read from the serial port
         with DEL.

       o Replace every CR character with CR and LF when echoing to the terminal
         (if local-echo is enabled).

gogonkt, 2017-03-06:
Hey Dear!

Well, I found that miniterm.py (a pySerial tool) work too
$ miniterm.py -e /dev/tty.wchusbserial1a20 115200
-e for local echo and it auto handle crlf
but no idea for screen or picocom

http://stackoverflow.com/questions/7812142/how-to-toggle-cr-lf-in-gnu-screen

zeroflag, 2017-03-06:
punyforth should work with both LF and CRFL.

I tried picocom with the following command:

picocom /dev/cu.wchusbserial40130 --b 115200 --omap crcrlf --imap crcrlf --echo

It works, although there are 2 crlf after each line, and it doesn't seem to be buffering the lines before sending it.

gogonkt, 2017-03-06:
wow, It works now. Great news!
I like miniterm solution now and It come with pySerial, very handy and cross OS.

zeroflag, 2017-03-06:
cool!

closed 2017-03-06 by zeroflag