zeroflag/punyforth #21: What is IP when using AP

RigTig, 2017-03-06:
Just after booting I run the following 4 commands:
`172 16 0 10 >ipv4 wifi-set-ip
4 3 0 AUTH_WPA2_PSK str: "1234567890" str: "AVAGO" wifi-softap
1 172 16 0 1 >ipv4 dhcpd-start
wifi-ip type`
and the result is 

> 0.0.0.0(stack)

Now the netcon-... commands depend on wifi-ip being correct, so I am having trouble connecting to this server from other machines. I've followed the code back to an RTOS call and all looks ok at the punyforth end. Is this another RTOS problem?

RigTig, 2017-03-08:
Aha! Maybe not RTOS. It seems that the get has the station mode hardcoded into the request, so getting the IP address in AP mode (or the AP address when in both station and AP) only returns the station address. Now if this is true, then it also seems the set has the index hardcoded to AP address. Anyway, I've pushed the limit of what I can do. Please check further.

In wifi.forth, 
`: wifi-ip ( -- str )
    here 16 allot
    16 over wifi-ip-str ;`

In ext.S,
`defprimitive "wifi-ip-str",11,wifi_ip_str,REGULAR
    DPOP a2                    // buffer
    DPOP a3                    // buffer size
    CCALL forth_wifi_get_ip_str
    NEXT`

In forth_wifi.c,
`void forth_wifi_get_ip_str(char * buffer, int size) {
    struct ip_info wifi_info;
    sdk_wifi_get_ip_info(0, &wifi_info);
    struct ip_addr ip = wifi_info.ip; 
    snprintf(buffer, size, IPSTR, IP2STR(&ip));`

Now, I cannot find the code for sdk_wifi_get_ip_info() but the nearest I get is Espressif document 20B-ESP8266_RTOS_SDK_API Reference.pdf which says
[image]

zeroflag, 2017-03-08:
thanks for the finding, I'm going to check this

RigTig, 2017-03-08:
It occurs to me that when in dual mode (STATIONAP_MODE) that there are two possible IPs to set and get, so it seems that those forth functions need to either have a parameter to say which IP or for there to be different words in forth.

zeroflag, 2017-03-13:
I introduced a new word:

```forth
: ap-ip ( -- str ) 1 ip ;
```

This works the same way as wifi-ip but gets the softap ip instead of the station ip. I also modified wifi-ip as follows:

```forth
: wifi-ip ( -- str ) 0 ip ;
```
I didn't rename wifi-ip to station-ip because of backward compatibility.

closed 2017-03-13 by zeroflag

RigTig, 2017-03-14:
Thank you for the new word. It has allowed me to poke around further. I have noticed that the softap configuration is not remembered by the ESP8266, so needs to be set up each boot. This might well be an RTOS issue or might just be that there is a flag somewhere to make the configuration stick. For my purposes, it is not an issue at this stage, since initialisation code is needed anyway. 

You might like to revisit the comment in readme.md which says

> The Wi-Fi settings are persistently stored by the ESP8266, there is no need to setup the Wi-Fi at every startup.

zeroflag, 2017-03-20:
Thanks, I updated the comment in readme.md.