zeroflag/punyforth #23: How know that dhcp client is running, so can use netcon words?

RigTig, 2017-03-21:
As part of startup code on an ESP12 I connect to a server (actually an ESP14) with:
`str: "1234567890" str: "AVAGO" wifi-connect`
and also start up a couple of tasks. One of the tasks needs to 
`PORT HOST TCP netcon-connect`
but the wifi-connect has not completed setting up the route so the netcon-connect aborts (usually -10).
One way around this is just to catch the error from netcon-connect and loop until it is successful, but it just does not look tidy. Is there a way to check the status of the dhcp client before attempting the netcon-connect?

zeroflag, 2017-03-23:
Maybe you could check if an ip address is associated (using wifi-ip, softap-ip) to the esp or not. But keep in mind that these words create and return a newly allocated string each time. So it's not a good idea to use them in a loop unless doing some extra memory housekeeping.

In the esp-open-rtos there is a function called sdk_wifi_station_get_connect_status().

https://github.com/SuperHouse/esp-open-rtos/blob/7c702d7f09f4f810144a17ae3d0ebd110ee77599/tests/cases/04_wifi_basic.c#L124

But currently this is not exposed as a forth word.

RigTig, 2017-03-23:
Thanks for info, zeroflag. I wondered if I had just not seen something. I think just doing a loop around the netcon-connect until it works is good enough for my purpose.

A related issue I am chasing right now is that netcon reads do not seem to time out. I can see a value being set in netcon-new, as follows:
`: netcon-new ( type -- netcon | throws:ENETCON ) override
    netcon-new
    dup 0= if ENETCON throw then 
    RECV_TIMEOUT_MSEC over netcon-set-recvtimeout ;
`
but I've had netcon-readln wait for hours - at least I think so. I am still struggling with keeping track of multiple tasks on 2 ESP8266s attempting to co-operate. My brain gets very full sometimes (most times?). Well, the co-operation part was not too hard, but handling comms failures is doing my head in. Sometimes just figuring out which end of the comms screwed up is hard and I spend hours poking around the wrong code before I get a clue. I keep learning, but sometimes there's a few steps back before I get to go forward again. 
Many thanks again for your help.

zeroflag, 2017-03-23:
> A related issue I am chasing right now is that netcon reads do not seem to time out.

This is quite possible and looks like a bug. The purpose of RECV_TIMEOUT_MSEC is not to block the running tasks (threads) by the one that waits for the data. The word read-ungreedy executes a pause (which yields control to an other task) when timeout occurs then retries forever. So RECV_TIMEOUT_MSEC is an internal timeout, but there should be an other timeout to be able to exit this loop. This is missing right now, thanks for letting me know.

I'm going to try to fix this in the weekends.

zeroflag, 2017-03-26:
I added option to specify global read timeots on netcon sockets.

```forth
10 READ_TIMEOUT_SEC ! \ set read timeout to 10 seconds
```

or

```forth
-1 READ_TIMEOUT_SEC ! \ -1 indicates no timeout
```

The netcon-read and netcon-readln throws ERTIMEOUT exception when a timeout occurs.

RigTig, 2017-03-28:
Many thanks for timeout. I'd missed reading the internal timeout in read-ungreedy. Time for me to have another go.

closed 2017-03-28 by RigTig

zeroflag, 2017-03-28:
Probably it would be better to be able to set the read timeout on individual connections instead of globally. So this may change in the future.

zeroflag, 2017-04-23:
Please note that I replaced the global timeout with an individual timeout per netcon.

\ words for setting and getting the socket level timeout

netcon-read-timeout! ( read-timeout-in-seconds netconn -- )
netcon-read-timeout@ ( -- read-timeout-in-seconds )

\ example

variable: socket
PORT HOST TCP netcon-connect socket ! 
60 socket @ netcon-read-timeout! ( set the timeout of this socket to 60 seconds )
socket @ netcon-read-timeout@ . cr ( print out the timeout )