zeroflag/punyforth #24: char:

GerardSontag, 2017-03-24:
char: is not an immediate word and this prevent writing it, easily, in a column word on the opposite hex: can be used in interactive and compiling mode.
Is it possible to have char: written in the same manner as hex: like this
: char: immediate word drop c@ interpret? invert if ['], , then ;
Yes i can overwrite it, but do you think it has to be include in the core?

zeroflag, 2017-03-26:
Yes I think it makes sense to work like this. Currently this is the typical usage of char: from inside a column  def.

```forth 
[ char: F ] literal
```

The [ temporary switches to interpretation mode, then char: parses the given character and finally literal compiles it. This is quite clumsy.

Using the way you suggested would make the usage easier. Let me check first if it causes any problem or incompatibility.

zeroflag, 2017-03-27:
I made it state aware as you suggested. This won't work if someone intentionally wants to force the interpretation semantics in compile mode, but I don't think this is a common usage (like ' and [']) of char:.

closed 2017-03-27 by zeroflag

GerardSontag, 2017-03-27:
Hello,
Thank you!
>> This won't work if someone intentionally wants to force the interpretation semantics in compile mode
This is nearly the same as hex: