zeroflag/punyforth #26: request to flush out getting started docs.

mhoffma, 2017-03-26:
On your page that describes setup you should also point out,(sorry if I missed it) you need to run modules.py and select a program to install as uber.forth.

serial connection requires local echo and crlf.

picocom $PORT -b 115200 --omap crlf --echo

BTW this is a really cool program very light weight.

zeroflag, 2017-03-26:
Yes, I agree that this need to be clarified a bit more as it confused other people too. Probably I'll write a separate page about this and link it to the readme.

zeroflag, 2017-03-27:
Hi,

I added a new page https://github.com/zeroflag/punyforth/wiki/Developing-Punyforth-applications

Let me know if you think there is anything missing.

closed 2017-03-27 by zeroflag

GerardSontag, 2017-03-28:
Hello,
May be you can add the ability to send files as soon as you have flashed punyforth.
I am using Tera Term as the terminal emulator and it let me send files by simply doing a drag n drop of the file. I've seen that Picocom let you send files too.
I other words you don't need to reflash your source as long as it's not working/perfect but in case of crash or reset you need to resend your file(s).
Add something like 
-mytest
marker: -mytest
at the beginning of your file and you're ready!

zeroflag, 2017-03-29:
Good point, I'll add this tool.

zeroflag, 2017-04-14:
I added a short section about sending files, and using the marker: word.

https://github.com/zeroflag/punyforth/wiki/Developing-Punyforth-applications