zeroflag/punyforth #27: I2C

BLavery, 2017-06-27:
I am trying to port the SPI example of oled SSD1306 to use i2c instead. I haven't succeeded yet. I am suspicious that this call:
**defprimitive "i2c-write-slave"**
at line 550 in ext.S of punyforth
is missing one of the 4 parameters of the corresponding C call 
**int i2c_slave_write()**
at line 120 in i2c.h at https://github.com/SuperHouse/esp-open-rtos/blob/master/extras/i2c/i2c.h

Am I missing something?
I do hope I don't have to set up to recompile punyforth, just for a 1-line change to ext.S

Brian

zeroflag, 2017-06-27:
Earlier i2c_slave_write had 3 parameters, see: https://github.com/SuperHouse/esp-open-rtos/commit/fd1081a0cc1e0f8723087b366aa81750f4638d1f#diff-938bcf30e93ecd659b5e0be1995b2c3d

The punyforth binary should be compiled against the earlier version of esp-open-rtos, so it should be the problem, but I'll double check it.

zeroflag, 2017-07-02:
Yes it has 3 parameters in that build, so this shouldn't be the problem. What kind of error do you get?

zeroflag, 2017-09-09:
I've just tested the i2c with a wemos oled shield and it worked fine. Closing this.

closed 2017-09-09 by zeroflag