zeroflag/punyforth #28: tcp-rpl "library"   , is hanging punyforth , what can be wrong

enriquepinedo, 2017-08-10:
Dear Attila,

I am triying to make  REPL  (tcp-rpl library )  function work. 

Punyforth  is finally comunicating fine trough the usb and I was able to test some GP-IO tasks without problems. 

Now I am following  your example of "Running punyforth from a power outlet" to redirect all in-out from the net , and be able to download new words from a wifi link. 
(please note I am doing all this from Windows, I only needed Linux to create the first "core.forth" , from them on I coninued flashing the ESP8266 from the windows flashing utility)

For this example( Sonoff "smart socket") , we need 
1)  gpio.forth \ ok, it did work then 
2)  tcp-repl.forth     which  apparently  requires to install first "netcon.forth" and "mailbox.forth" 

My problems seem to be with    "netcon.forth" : 

we have this problem that will not compile see (**) 

: netcon-new 
( type -- netcon | throws:ENETCON ) override netcon-new     \ <--- ** this is an autoreference ? 
dup 0= if ENETCON throw then RECV_TIMEOUT_MSEC over netcon-set-recvtimeout ; 

I really do not know were the problem is in netcon.forth 
but, when I try to install those needed files to get tcp-repl working , 
it allways hangs the system.  I get into infinte resets and reboots. 

 I am thanking in advance for your help. 

Regards Enrique

**  can I comment out  "override  netcon-new"  ?

enriquepinedo, 2017-08-10:
This are the messages  I get  ... (it makes no sense, since I have downloaded all the dependencies, 
gpio, wifi-forth ,  tcp-rcpl  forth, and anything in between....

pm_task_hdl : 3fff8758, prio:1, stack:176
frc2_timer_task_hdl:3fffb328, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ May 18 2016 18:24:52
phy ver: 273, pp ver: 8.3

Punyforth loading..
mode : sta(18:fe:34:e1:2e:f7)
add if0
scandone
add 0
aid 5
cnt 

connected with TP-LINK_E565B8, channel 4
dhcp client start...
ip:192.168.1.102,mask:255.255.255.0,gw:192.168.1.1

Undefined word: mailbox:

Undefined word: task:

Undefined word: task:

Undefined word: activate

Undefined word: netcon-tcp-server

Undefined word: connections

Undefined word: deactivate

Undefined word: command-loop

Undefined word: netcon-readln

Undefined word: activate

Undefined word: connections

Undefined word: command-loop

Undefined word: netcon-dispose

Undefined word: deactivate

Undefined word: repl-start

Undefined word: multi

Undefined word: repl-server-task

Undefined word: repl-worker-task

Undefined word: traceback
Fatal exception (29): 
epc1=0x40239a18
epc2=0x00000000
epc3=0x4023980b
excvaddr=0x00000008
depc=0x00000000
excsave1=0x40213374
Registers:
a0 40213374 a1 3fffbe70 a2  0000000a a3  3fffc318
a4  00000009 a5  000000fd a6  0000007f a7  60000000
a8  00000008 a9  3ffe920c a10 ffffffff a11 3ffe882c
a12 00000000 a13 3ffee9f8 SAR 0000001e

Stack: SP=0x3fffbe70
0x3fffbe70: 00000000 00000000 00000000 00000000
0x3fffbe80: 00000000 00000000 00000000 a5a5a5a5
0x3fffbe90: e12edb34 0000005c 3fffbe20 e1815fed
0x3fffbea0: 3fff7128 3fff7128 3fffbe98 3fff7120
0x3fffbeb0: 0000000d ec31493c e550588b 3fffbe98
0x3fffbec0: 00000000 00000002 3fffb490 796e7570
0x3fffbed0: 74726f66 7ef00068 0077dd48 00000002
0x3fffbee0: 00000000 00000000 b0a7c100 04b2780e

Free Heap: 12712
_heap_start 0x3fff78e8 brk 0x3fffcc44 supervisor sp 0x3ffffb00 sp-brk 11964 bytes
arena (total_size) 21340 fordblks (free_size) 748 uordblocks (used_size) 20592

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

zeroflag, 2017-08-10:
Hi Enrique,

Could you describe how excatly do you flash the code to the esp step by step? The error messages indicates that the _mailbox:_ word which is located in mailbox.forth and requires by netcon is not defined. If you use the tools and scipts like modules.py, flash.sh/flash.bat correctly then this should not happen.

This is correct forth code, netcon-new is defined in terms of its older definition. See the _override_ word in the readme.

```forth
: netcon-new ( type -- netcon | throws:ENETCON ) override 
  netcon-new dup 0= if ENETCON throw then 
  RECV_TIMEOUT_MSEC over netcon-set-recvtimeout ;
```

enriquepinedo, 2017-08-11:
Hi Attila ! 
I did find what was wrong.  I am still learning punyforth differences... because your forth has some words
that I do not know and are not standard F83.  F. ex. override, so I thought in first
case it was some corrupted file I was compiling in tcp-rcpl  . Ok now I see override  is 
an immediate word.  
I was also confused looking at the word  "words" , I did even lost a lot of time compiling one for myself
it took a lot of time and was also running wrong and throwing trash in the stack, all without knowing 
that   "words"   was allready there and you called it  "help " ! 
I am still in the transition phase , so please have a bit of patience with my questions. I think I will have
some more....

I went step by step looking at all the code was missing, and found all the "libraries" 
wifi  ringbuff  netcon  tasks.... etc etc  I placed everything together by hand, (remember
I am working from Windows)  so I could  finally  assemble a new ueber.forth and voila !
it did work ! No definition  was missing and Punyforth boots normal.

Of course this is all my fault because I am not using Linux, it could be overcomed with 
the python routines you created.

By the way I am also admiring your excellent work with punyforth.
now I am triying to establish connection with Putty... it looks promising.

Best regards
Enrique

zeroflag, 2017-08-11:
Cool, I'm glad you make it work.

I regularly flash the esp on Windows and it's really not any different or more complicated than doing the same on Linux or Mac.

Open a cmd.exe

```bash
cd punyforth/arch/esp8266/bin/
python modules.py tcp-repl
flash COM3
```

and that's it. If you want to flash some additional custom code you can use the following

```bash
python modules.py --app myapp.forth tcp-repl
flash COM3
```

(assuming the COM port is COM3)

closed 2017-08-11 by zeroflag