zeroflag/punyforth #29: need help to convert numbers to string

enriquepinedo, 2017-08-21:
Dear Attila, 

would you help me  formatting  numbers to strings ?
    
I am missing   :      <#    # #  #   #s  #> 

how is this implemented in punyforth ?

Best regards

zeroflag, 2017-08-21:
There is a >str word for converting numbers to strings but no formatting is supported at this moment.

```forth
(stack) here dup 1234567 >str
(stack 134534064) type cr
1234567
(stack) 
```

enriquepinedo, 2017-08-22:
Hi Attila !  Thank you again it did work well.

I have some toughts....

1) why not a punyforth forum ?
2) I was looking for  a "see " word  and then I found your conversation with holinepu in the  punyforth lobby  about  assembler disassembler for punyforth , is it possible to get a copy of that file. Apparently
the link is not working any more. 
3) What about adding more tools for punyforth to run in Win32forth in the future ? 
I had particullary troubles (I think this was caused by complicated file   paths on my PC. ) f.e.
I had problems with the installation of python to make punyforth start, so I did it all in a VM under Ubuntu, and from there It was straightforward. Also I wanted to try Craig´s java based  file loader but I could not run the java files in my Windows machine. It was too complicated for me (sory for that). Again I think this all depends on installation problem and file paths, that do not match.  But if all tools would be using  Win32forth as a platform it would be super easy to follow in any machine. 
  
Best regards

\   I keep practising and will ask for your kind help when I have more  doubts...

I would

zeroflag, 2017-08-23:
A forum seems to be a bit overkill to me (but maybe using some existing infrastructure, like creating a subreddit on reddit.com would make sense). 

Here is a gist about the disassembler

https://gist.github.com/zeroflag/1dfad97e131b7ad14a5b77d5af49ac7b

I still need to dive into this and understand but I didn't have the time revently.

enriquepinedo, 2017-08-25:
Thank you Attila !

I will study this disassembler.

closed 2017-09-10 by enriquepinedo