zeroflag/punyforth #30: example of  i2c  words

enriquepinedo, 2017-09-10:
Hi Atila !  

I am having troubles finding an example of  i2c use, I am triying a PCF8574 lcd
and saw yours 
 i2c-write-slave 
i2c-read-slave
 i2c-write 
i2c-read 
i2c-init
 i2c-stop
 i2c-start 

my questions are  how to adress the chip , stack  notation for each word, etc.

is it Ok to use  D1 SCL  and D2 SDA, must those Pins be defined beforehand   to the
ESP8266 or are they  fixeded pins  ?    lots of doubts....  

Can you help me ?

Best regards, 
Enrique.

zeroflag, 2017-09-10:
Here is an example that sends a 8 byte buffer to a slave with 3C address.

```forth
5 ( D1 ) constant: SCL
4 ( D2 ) constant: SDA
8 buffer: data
\ .. fill data ..
SDA SCL i2c-init
8 data 16r3C i2c-write-slave
```

zeroflag, 2017-09-10:
Also I started working on ssd1306 i2c display driver that can serve as an example.

This one is already usable but few things might change in the future.

https://github.com/zeroflag/punyforth/blob/master/arch/esp8266/forth/ssd1306-64x48-i2c.forth

enriquepinedo, 2017-09-10:
Excellent Atila !

I will look at it and experiment on that, I keep you posted.
thanks again for bewying helpful and  good work !
Enrique

closed 2017-09-13 by enriquepinedo