zeroflag/punyforth #31: Need SPI documentation

CraigLindley, 2017-09-11:
I'm trying to code up an spi-read function. I guess it needs to be implemented using the spi-send function but there is no documentation on how the function works. Also I would like to know how to set the spi speed correctly for the spi-init function. 

If you can either provide me with an spi-read function or point me at some documentation so I can code one myself I would appreciate it.

zeroflag, 2017-09-11:
The spi functions are based on 

There is a thin wrapper:

https://github.com/zeroflag/punyforth/blob/feab7a0fea1c815a5dc441085251fdea44c7089e/arch/esp8266/rtos/user/forth_spi.c

And the Forth words are here:

https://github.com/zeroflag/punyforth/blob/6a2e89c1969c12798d627f9b5c49326139f99e8a/arch/esp8266/ext.S#L394

There isn't any user friendly documentation yet, but you can check the stack usage. 

For example spi-send8 is defined as follows:

```forth
defprimitive "spi-send8",9,spi_send8,REGULAR
    DPOP a2                    // bus
    DPOP a3                    // data
    CCALL forth_spi_send8
    DPUSH a2
    NEXT
```
So the stack effect should be

```forth
: spi-send8 ( data bus -- result )
```

I think you're right saying that the read should be implemented by using the send. But I didn't need to use read so far.

There are some examples of using spi-init and send in 

https://github.com/zeroflag/punyforth/blob/master/arch/esp8266/forth/ssd1306-spi.forth

CraigLindley, 2017-09-11:
> ea44c7089e/arch/esp8266/rtos/user/forth_spi.c
> 6139f99e8a/arch/esp8266/ext.S#L394
> esp8266/forth/ssd1306-spi.forth

Hi Attila

Yes I've investigated all of that over the last couple of days but it
doesn't explain how to use spi-send for example. There is an indata and an
outdata pointer but in reference to what? Is indata the data read from the
spi interface or the data being sent out of the interface? Also, does one
have to pass 0xFF into the function for each byte that is being read?

The low level functions that the forth words call must be documented
somewhere.

Also in the spi-init function has a freq-div parameter. It it set using
your function

5 2 spi-get-freq-div constant: SPI_FREQ_DIV_8M

does this constant divide the ESP8266 clock by 8 million or does it provide
an 8 Meg clock for spi?

If you get a chance to write an spi-read function I would love to see it.

BTW, I'm trying to use Punyforth to build an Internet radio. I will give
you this code if and when I get it working.

Craig Lindley

-- 
Craig Lindley / Heather Hubbard

495's Recordings: craigandheather.net/495spage.html
New Recordings: craigandheather.net/cnmpage.html
Latest rock CD: craigandheather.net/oneinarow.html
Latest non-rock CD: craigandheather.net/craigdoesfingerstyle.html

Personal Website: craigandheather.net

Phone: (719) 495-1873
Cell: (719) 502-7925

If you’re one in a million, there are now seven thousand people exactly
like you.

zeroflag, 2017-09-11:
Here is the documentation of spi_transfer from https://github.com/SuperHouse/esp-open-rtos/blob/61c3d509e5b930c6c1b269ceff99e2d1f60d2010/core/include/esp/spi.h#L275

```
 * \param bus Bus ID: 0 - system, 1 - user
 * \param out_data Data to send.
 * \param in_data Receive buffer. If NULL, received data will be lost.
 * \param len Buffer size in words
 * \param word_size Size of the word
 * \return Transmitted/received words count
 */
size_t spi_transfer(uint8_t bus, const void *out_data, void *in_data, size_t len, spi_word_size_t word_size);
```

So the out_data buffer contains the data to be sent and the other one is the receive buffer. I didn't need the receive buffer when I used this (I just passed 0) so I don't have more information on this beside what's in the C comment.

This may help regarding the spi-get-freq-div constant.

```
/**
 * Macro for use with spi_init and spi_set_frequency_div.
 * SPI frequency = 80000000 / divider / count
 * dvider must be in 1..8192 and count in 1..64
 */
#define SPI_GET_FREQ_DIV(divider, count) (((count) << 16) | ((divider) & 0xffff))

/**
 * Predefinded SPI frequency dividers
 */
#define SPI_FREQ_DIV_125K SPI_GET_FREQ_DIV(64, 10) ///< 125kHz
#define SPI_FREQ_DIV_250K SPI_GET_FREQ_DIV(32, 10) ///< 250kHz
#define SPI_FREQ_DIV_500K SPI_GET_FREQ_DIV(16, 10) ///< 500kHz
#define SPI_FREQ_DIV_1M   SPI_GET_FREQ_DIV(8, 10)  ///< 1MHz
#define SPI_FREQ_DIV_2M   SPI_GET_FREQ_DIV(4, 10)  ///< 2MHz
#define SPI_FREQ_DIV_4M   SPI_GET_FREQ_DIV(2, 10)  ///< 4MHz
#define SPI_FREQ_DIV_8M   SPI_GET_FREQ_DIV(5,  2)  ///< 8MHz
#define SPI_FREQ_DIV_10M  SPI_GET_FREQ_DIV(4,  2)  ///< 10MHz
#define SPI_FREQ_DIV_20M  SPI_GET_FREQ_DIV(2,  2)  ///< 20MHz
#define SPI_FREQ_DIV_40M  SPI_GET_FREQ_DIV(1,  2)  ///< 40MHz
#define SPI_FREQ_DIV_80M  SPI_GET_FREQ_DIV(1,  1)  ///< 80MHz

```

CraigLindley, 2017-09-11:
> 61c3d509e5b930c6c1b269ceff99e2d1f60d2010/core/include/esp/spi.h#L275

Thanks this helps. I'll try to write my spi-read8 function using this
information.

-- 
Craig Lindley / Heather Hubbard

495's Recordings: craigandheather.net/495spage.html
New Recordings: craigandheather.net/cnmpage.html
Latest rock CD: craigandheather.net/oneinarow.html
Latest non-rock CD: craigandheather.net/craigdoesfingerstyle.html

Personal Website: craigandheather.net

Phone: (719) 495-1873
Cell: (719) 502-7925

If you’re one in a million, there are now seven thousand people exactly
like you.