zeroflag/punyforth #32: Need way to print integers in hex

CraigLindley, 2017-09-12:
Since Punyforth doesn't have words like: base and hex how does one print out numbers in hexidecimal? I am working with hardware where I need to display values during debugging in hex. Is there an easy way to do this or do I have to write my own decimal to hex string conversion?

zeroflag, 2017-09-13:
There isn't anything builtin at this moment. I afraid you have to write your own converter.

CraigLindley, 2017-09-13:
I did and here it is if you want/need it

marker: _HEXPRT_

variable: jIndex
15 constant: hexstrsize

hexstrsize byte-array: hexstr

\ Print an integer in hex
: hexprt ( n -- )

  \ Clear the hex string array
  hexstrsize 0 do 0 i hexstr c! loop

  \ Initial index to 0
  0 jIndex !

  begin
    dup                       ( n -- n n )
    0<>                     ( n n -- n f )
  while                     ( n f -- n )
    dup                       ( n -- n n )
    16 %                    ( n n -- n rem )
    dup                   ( n rem -- n rem rem )
    10 <              ( n rem rem -- n rem f )
    if                  ( n rem f -- n rem)
      48 +                ( n rem -- n rem+48 )
    else                        ( -- n rem )
      55 +                ( n rem -- n rem+55 )
    then
    jIndex @ hexstr     ( n rem+n -- n rem+n addr )
    c!             ( n rem+n addr -- n )
    1 jIndex +!               ( n -- n )
    16 /                      ( n -- n n/16 )
  repeat
  drop                        ( n -- )

  \ Read out the string backwards
  0 hexstrsize 1- do i hexstr c@ emit -1 +loop
;

: hexprtln ( n -- )
  hexprt cr
;

-- 
Craig Lindley / Heather Hubbard

495's Recordings: craigandheather.net/495spage.html
New Recordings: craigandheather.net/cnmpage.html
Latest rock CD: craigandheather.net/oneinarow.html
Latest non-rock CD: craigandheather.net/craigdoesfingerstyle.html

Personal Website: craigandheather.net

Phone: (719) 495-1873
Cell: (719) 502-7925

If you’re one in a million, there are now seven thousand people exactly
like you.

zeroflag, 2017-09-13:
Thanks for sharing it.

closed 2017-09-13 by zeroflag