zeroflag/punyforth #33: Esplorer works also well with Punyforth

enriquepinedo, 2017-09-14:
I only wanted to share my experience with Punyforth and Esplorer, I can edit side by side, 
send 1 line of code, a selection of code or a complete file , very easy ! 
It would be useful  to find a way  to edit the buttons of esplorer,  to have shortcuts snippets etc at hand....
like it does for lua and  mphyton, something I could not find how to modify. 
I am enjoying very much Punyforth, thanks Attila !

enriquepinedo, 2017-09-14:
I ve submitted a picture of esplorer and punyforth it does not show up ???

zeroflag, 2017-09-16:
I'm happy to hear it works well with Punyforth. I don't know how can you modify it but you can fork their repo, send pull request to them, etc.

I guess this is github repo of Esplorer

https://github.com/4refr0nt/ESPlorer

closed 2017-09-16 by zeroflag