zeroflag/punyforth #34: missing further examples  of turnkey and save

enriquepinedo, 2017-09-16:
Dear Attila, 
This is the only example we have of turnkey from the dt data logger
and there is a flash forth  with another  turnkey definition and save-loader
***********from data-logger *************
: main ( -- )         log-measurement          600000000 deep-sleep ;
   ' boot is: main         turnkey abort
********************************************
from falsh.forth
: save-loader ( -- )         0 block drop c        0 row dup heap-size >str dup strlen +        
    str: ' heap-start ' over 12 cmove             12 + dup dst >str dup strlen +        
    str: ' read-flash drop boot' swap 22 cmove          update flush ;
*****************************************

I triyed to save a forth incrementally (without using any external flash memory) bare bone
esp8266 chip.  And I failed, it will all the way fall in reboot.  It happend the same with 
turnkey....
I am requesting kindly  more examples or explanations  of both words
and also if there is a way to interact with punyforth test and compile some words incrementally
and then make a "SAVE"  so the system will preserve the actual dictionary  for the next boot
cycle  for future use and interaction....   do you understand what I mean ?  
save would make the same as turnkey except for executing the ' word stored at boot, it will
coolstart from the interpreter prompt.

(I am enyoing very much your work, long life to Punyforth !   :)

GerardSontag, 2017-09-16:
Enrique,
Please reread https://github.com/zeroflag/punyforth/wiki/Developing-and-deploying-Punyforth-applications
and try the example.
You just need to generate a new uber.forth for your application.
Hope this help you.

zeroflag, 2017-09-16:
Turnkey is an ongoing expermiment and it is meant to be an _optimalization_ that speeds up startup time. This is only important if you try to power the esp using batteries (with deep sleep) and you don't want to waste power during startup. When you execute turnkey then it'll save the current state of the dictionary in _binary_ form to the flash and it'll also insert a small loading code that reads the dictionary upon startup.

Usually you don't need this. Just follow the link what @GerardSontag mentioned there is the recommended way to save code permanently to the flash.

enriquepinedo, 2017-09-16:
Thank you Gerard and Attila !

Yes Gerard of course I am generating a new Ueber-forth with all associated  dependencies, 
(I do this by hand selecting all files, that I copy to the new Ueberforth. This part is not the problem). 

The problem arises when I am loading the example  of  **flash.forth**   to create the new  Ueberforth, 
Punyforth repeatedly reboots It seems there  is an error in the code  of  **flash.forth**, please if you 
can check that  for me :

0 constant: FLASH_OK
1 constant: FLASH_ERR
2 constant: FLASH_TIMEOUT
3 constant: FLASH_UNKNOWN

( blocks )

exception: EBLOCK
4096       constant:      SIZE
hex: 51000 init-variable: block0
FALSE      init-variable: dirty
SIZE       buffer:        buf
variable:  offs

: check-err ( code -- | EBLOCK )
    dup FLASH_OK <> if
        print: 'SPI FLASH ERROR: ' . cr
        EBLOCK throw 
    then 
    drop ;

: >sector ( block# -- sector# ) SIZE / ;
    
: flush ( -- )
    dirty @ if
        offs @ >sector erase-flash  check-err
        SIZE buf offs @ write-flash check-err
        FALSE dirty !
    then ;
    
: block ( block# -- addr )
    flush
    SIZE * block0 @ + offs !
    SIZE buf offs @ read-flash check-err
    FALSE dirty !
    buf ;
    
: update ( -- ) TRUE dirty ! ;

( screen editor )

128 constant: COLS
32  constant: ROWS

: row   ( y -- addr )   COLS * buf + ;
: ch    ( y x -- addr ) swap row + ;
: type# ( y -- )        dup 10 < if space then . space ;
    
: list ( block# -- )
    block drop
    ROWS 0 do
        i type#
        i row COLS type-counted
    loop ;

\ editor command: blank row
: b ( y -- )
    COLS 2 - 0 do 32 over i ch c! loop
    13 over COLS 2 - ch c!
    10 swap COLS 1-  ch c!
    update ;
    
: copy-row ( dst-y src-y -- )
    COLS 0 do
        2dup i ch c@ swap i ch c!
    loop
    2drop ;
    
\ editor command: delete row
: d ( y -- )
    ROWS 1- swap do i i 1+ copy-row loop
    ROWS 1- b ;

\ editor command: clear screen    
: c ( -- ) ROWS 0 do i b loop ;

\ editor command: overwrite row
: r: ( y "line" -- )
    dup b row
    begin
        key dup line-break? invert
    while
        over c! 1+
    repeat
    2drop ;

\ editor command: prepends empty row before the given y
: p ( y -- )
    dup ROWS 1- do
        i i 1- copy-row
    -1 +loop
    b ;

 
defer: boot    
: dst ( -- n ) block0 @ SIZE + ;
: heap-size ( -- n ) usedmem align ;
: save-loader ( -- )
    0 block drop c
    0 row dup heap-size >str dup strlen +        
    str: ' heap-start ' over 12 cmove
    12 + dup dst >str dup strlen +        
    str: ' read-flash drop boot' swap 22 cmove 
    update flush ;
        
: turnkey ( -- )
    heap-size SIZE / heap-size SIZE % 0> abs + 0 do
        dst >sector i + erase-flash check-err
    loop
    heap-size heap-start dst write-flash check-err 
    save-loader ;

enriquepinedo, 2017-09-16:
Hi Attila, I am sending line by line to debug and see what can be wrong, is it possible I am getting out of memory ?

I could advance up to  "copy-row"  definition, then the system sayd "fatal exception 9"
this is the dump of the screen:

(stack 1073673520) .. Fatal exception (9): 
epc1=0x4023þþs2
epc2=0x       0
epc3=0x402033þo
excvaddr=0x4010434s
depc=0x       0
excsave1=0x402133$4
Registers:
a0 402133$4 a1 3sssf$0 a-2d        2
a-2d        3
a-2d        4a-2d        54a-2d        6ha-2d        $a-2d        áOa-2d        þða-2d        ?ÿa-2d         a-2d        o

zeroflag, 2017-09-17:
@enriquepinedo, you can check the available free dictionary space by typing

`forth
freemem . cr
```

It's possible you've ran out of free space as the available dictionary space is fairly limited (about 24k).

closed 2017-09-17 by enriquepinedo