zeroflag/punyforth #35: request for Memory map of Punyforth

enriquepinedo, 2017-09-17:
Dear Attila,

Sory if this is explained elsewhere I did not found it

I am very confused without a memory map of Punyforth, I need to understand
what is happening with my reboot problems I am experimenting with
different programs. It is also almost impossible to advance with Forth 
without having a memory map of the full system, I feel like shooting
with a machinegun blindfolded ! It is tough.

I am copying below an example of a memory Map of MicroPhyton for ESP8266
please if you can edit it and correct it for Punyforth to help
everybody visualize better how is Flash and Ram used in Punyforth, and
were is our free space going after loading dependencies. 
The example of the internal of  Micropython on ESP8266 
helped me  understand a bit further how the Rtos and Wifi will
consume memory  etc... 

I have many questions left : 1)  is Uber.forth  at $51000
copyed from flash to RAM before execution ?  because it does not make
sense if we have 1 MB Flash , that 96 K or 64 K ram of ESP get so fast
gone !  In my application  1 LCD with i2cbus and 1 wifi Repl , it consumes
all my memory, I have  left only 2K ! After that I can almost not add
anything more  without falling into constant reboots. 
2) how can this 1MB or 4MB of internal flash been better used ? Is it now
a wasted space ?
3) I suppose there is no way to add an external SD card, this would not
help in any sense.  What else can be done ?

Mphyton Memory MAP :

-----------------------

-----------------------  0X4030_0000

MEMORY MAPPED FLASH ROM (1M)

-----------------------  0X4020_0000                       ]
                                                                           ]
FLASH MEMORY MAPPING CACHE #1              ]    6
16 K INSTARAM                                                 ]     4 
-------------------------0X4010_C000                     ]
                                                                           ]    K 
FLASH MEMORY MAPPING CACHE #0              ]
16 K INSTARAM                                                  ]
                                                                           ]
------------------------ 0X4011_0000                      ] 

FAST CODE INSTARAM (32K)

------------------------ 0X4010_0000

------------------------ 0X4001_0000

     BOOT ROM  (64)

------------------------ 0X4000_0000       ]
                                                            ]    9 
                                                            ] 
  SYSTEM DATA RAM  (16K)               ]   6
                                                            ] 
------------------------ 0X4001_0000       ]    
                                                            ]   K
                                                            ]
  USER DATA RAM (80 K)                    ]   
                                                             ]
------------------------ 0X3FFE_8000       ]

------------------------

Kind regards
Enrique

enriquepinedo, 2017-09-17:
This was the picture of the memory map, I can not add it as such on my message....
https://ksr-ugc.imgix.net/assets/005/420/685/7942660fd0259ae699e373b25c5c6fec_original.png?w=639&fit=max&v=1456359366&auto=format&lossless=true&s=37936c9f7731bedc214afb9b94c7d455

enriquepinedo, 2017-09-18:
I continue exploring punyforth and I have tests and results 
( all numbers below  in decimal.) 

core.forth    alone  gives   freemem   of                    16.776 bytes  free
on an empty system       here  points to         1073.653.560 ( i can not find any correlation
with physical memory in this  large number ) 

tcp-repl with all it dependencies and 
a bit of  code for an  i2c     LCD                 gives me     2.200 bytes freemem  
for programming
anything else I try to include after that  Punyforth will hang. I can only add
very elementary forth operands. after a "screen"  or a "block"  of txt (1024 k
text code)_   it will  hang.

I have been looking at  flash.forth dependencies, it has  only 1  needs "core.forth"

Questions :  is this editor working ?  It seems it saves to Flash memory, like I previously asked 
if there is a way to "SAVE" a system interactively and incrementally... 

I will try to understand more.....  apparently having Rtos and open-boot in the system
those calls are causing the lack of program memory, ( even when  ESP8266 has such a 
large ram  in terms of  a forth system). 

Kind regards
Enrique

zeroflag, 2017-09-18:
> I have many questions left : 1) is Uber.forth at $51000 copyed from flash to RAM before execution 

No, the text is not copied into the RAM. Punyforth compiles indirect thread code from the text and fills up the dictionary.

> I have left only 2K 

This sounds realistic. The _freemem_ word shows the available forth dictionary space, not the free system memory. There is an _osfreemem_ word for getting the free memory from RTOS. Punyforth allocates about 24k dictionary space which means that your program and its dependencies consume about 22k. 

When I was experimenting with larger dictionary spaces in the past, the system became unstable because not enough memory was left for RTOS. It usually crashed inside the lwip (tcp stack) code when it tried to allocate netcon buffers. So I decided to allocate only 24 forth heap, and leave the rest to RTOS.

You can experiment with larger memory settings by changing this line and recompile Punyforth

https://github.com/zeroflag/punyforth/blob/master/generic/data.S#L32

If you don't use the wifi and the tcp stack then it'll probably Ok.

But an easier way to fix this is to remove unused words from the uber.forth. 

> Questions : is this editor working ? It seems it saves to Flash memory, like I previously asked
if there is a way to "SAVE" a system interactively and incrementally...

There is an experimental editor in flash.forth but you need to use the --block-format parameter of modules.py when generating uber.forth.

enriquepinedo, 2017-09-19:
Ok excellent !  
Line     32      .........          .space 25152  

In your opinion  if I place   .space  to  50000   will it work ?  30000, 35000 ?  which are the limits ?

Can we expect in future a memory map of punyforth within the ESP8266 ?  I would like to help
in anything related to documentation to your proyect if you whish.
I am preparing for my own use a Spreadsheet with all the words of punyforth, like a glossary, the
old systems used to have , with stack notation and if possible or needed for more complex words
an example below so anybody can learn faster. As soon I have it ready I will pass this to you.
Thanks again for your time !

zeroflag, 2017-09-20:
> In your opinion if I place .space to 50000 will it work ? 30000, 35000 ? which are the limits ?

As far as I remember 30k should work but it will become unstable if you use the WiFi a lot.

> Can we expect in future a memory map of punyforth within the ESP8266 ?

Punyforth is not a bare metal Forth (at least now currently) but it runs as an RTOS task. This task is created by calling xTaskCreate, which allocates some memory dynamically. But I don't know how the memory allocation is handled internally.

> I am preparing for my own use a Spreadsheet with all the words of punyforth, like a glossary

That's great. I'm sure this will be useful for others too.

enriquepinedo, 2017-09-22:
Thanks Attila ! 
I will try this and let you know.

zeroflag, 2017-10-05:
FYI I made some optimalization in variable: and constant: that reduces their memory consumption. This should buy you some additional memory.

enriquepinedo, 2017-10-06:
Excelent !  THank you very much !

GerardSontag, 2017-10-08:
Hello Attila,
If you want to reduce the code size you can consider moving to a version with TOS in register. The a2 register is the best candidate for TOS because it is the first parameter when calling OS function. A lot of these calls when moved to this new version will end with 
DPOP a2
NEXT 
sequence which can be replace by a:
B PopNext
and 
somewhere 
PopNext: 
DPOP a2
NEXT
And for a lot of primitives you will save the final DPUSH.

zeroflag, 2017-10-08:
Hi Gerard,

I was thinking about TOS to register optimization earlier (mainly because of speed) but decided not to do it because some of the caveats. There are words like sp@ and sp! which would be difficult to use if TOS was in a register. Also one can write a code that treats the stack as an array which would likely break with TOS optimization. Task switching and exception handling would likely be more complicated as well.
And I'm not worried about the size of the compiled primitives as much because they're not stored in the same (~24K) dictionary space like the user defined words.

Attila

GerardSontag, 2017-10-09:
Hello Attila,
sp@ sp! will become:

defprimitive "sp@",3,spat,REGULAR
DPUSH		a2
mov			a2, a15
NEXT

defprimitive "sp!",3,spstore,REGULAR
mov			a15, a2
DPOP		a2
NEXT

Gérard

zeroflag, 2017-10-09:
Right. I just don't think the extra complexity would worth the effort. But if you think so feel free to send a PR with the changes required. I'm not against it, I know other Forths do this but mainly because of speed (which is not the main concern) and not memory usage.

GerardSontag, 2017-10-12:
Hello Attila,
I have done this work some months ago but at that time was unable to assemble the source (some weird things in ubuntu under w10) and know I am unable to merge in Github...
I may be give a try again by bypassing github.
BTW do you know good documentation about the ESP8266 I got the Xtensa Instruction Set Architecture but i miss some documentation about how 8266 is implementing over LX106
Gérard

zeroflag, 2017-10-25:
@GerardSontag no I don't any unfortunately. But there are some projects in github like the esp port of microphython or forthright which can be used. as an example. The API of FreeRTOS is well documented (http://www.freertos.org/a00106.html).

holinepu, 2018-08-13:
It seems that the flash rom is so big enough to add more compiled code into it. So if someone can design the metacompiler to append the code to it that would be great, right? I'll try, and you.

zeroflag, 2018-08-27:
Hi holinepu,

Could you elaborate the problem with create does>? I'm trying to recall how it works.

create: always creates a new word with an [enterdoes](https://github.com/zeroflag/punyforth/blob/653ef3c05c2efe635a079647326b0d8b3e9d25dd/arch/esp8266/primitives.S#L293) codeword/entrypoint instead of entercol (which is the usual entrypoint for colon definitions). 

The enterdoes expects a behaviour pointer (which is an execution token of an other word) after the codeword and some arbitrary data after this. It'll push the location of the data part onto the stack and invoke the behaviour word. The default behaviour word is nop, but does> overwrites it with a user supplied word.

```forth
: create: createheader enterdoes , ['] nop cell + , ; \ default behaviour is nop, does> overwrites this
```

Here you can see that it creates a dictionary header and inserts the enterdoes codeword as an entry point. After that it inserts the xt of nop. The data part is not initialized.

```forth
: does> r> lastword link>body ! ;
```

does> replaces the behaviour part with the user supplied word (from the return stack).

So a constant for example can be defined:

```forth
: constant: cerate: , does> @ ;
```

But punyforth uses distinct codewords for variable/constants to save some space.

holinepu, 2018-08-28:
Dear Attila,
  Thanks for the reply.
  I've been trying to compile the program in the rom area. It is succefull to do now.
But still we have to solve the problem of variables residing in rom/ram area.
The nop instruction is just to provide a ( random ) address in RAM where is a safe place.
Why not just give it a fixed RAM address ( 3FFFF800, for example )?
I'd like to put create: and does> in flash rom ( or ram ), that way there is no need nop accompaning to the twos.

variable: can be defined in rom area too, so long as it can provide a number which is a ram address.

  :  variable.base.address      hex: 3FFFF800                ;    \  3FFFF800 can be changed according to need.
  :  ram.address.replacing.nop  variable.base.address
  :  variable.counter           variable.base.address  4     ;    \  0  variable.counter  !  <-- how to intialize it?
  :  variable.area              variable.base.address  8  +  ;
  :  variable.area.wide         hex:                   400   ;    \      400  can be changed according to need.
  :  var(ram)  ( -- )

     createheader  hex:  40239B2C ,         variable.area     variable.counter  @   +  ['],  ,
     4  variable.counter  +!  hex: 402182C8 ( exit )  ,   ;
  :  create:(ram)    var(ram)   -4     variable.counter  +!   ;
  :  allot(ram)  ( -- n )  align  variable.counter  +!   ;

var(ram)  v1
var(ram)  v2
var(ram)  v3

see v1   see v2   see v3

\ ******************************************************************************************
  ( 3FFE9098  3FFE9088 )
  \ 3FFE909C  07 63 72 65 61 74 65 2F
                             :  create/
\ ------------------------------------------------------------------------------------------
  ( 3FFE90A4  40239B2C )                                 ( docolon )
  ( 3FFE90A8  40218E0C )                                 createheader
  ( 3FFE90AC  40219660 )                                 enterdoes
  ( 3FFE90B0  40218AB4 )                                 ,
  ( 3FFE90B4  4021824C )                                 ( dolit )  3FFE9090
  ( 3FFE90BC  40218A9C )                                 cell
  ( 3FFE90C0  40218174 )                                 +
  ( 3FFE90C4  40218AB4 )                                 ,
  ( 3FFE90C8  402182C8 )                                 exit

 see create:
\ ******************************************************************************************
  ( 40279098  40279088 )
  \ 4027909C  07 63 72 65 61 74 65 3A
                             :  create:
\ ------------------------------------------------------------------------------------------
  ( 402790A4  40239B2C )                                 ( docolon )
  ( 402790A8  40218E0C )                                 createheader
  ( 402790AC  40219660 )                                 enterdoes
  ( 402790B0  40218AB4 )                                 ,
  ( 402790B4  4021824C )                                 ( dolit )  3FFFF800
  ( 402790BC  40218A9C )                                 cell
  ( 402790C0  40218174 )                                 +
  ( 402790C4  40218AB4 )                                 ,
  ( 402790C8  402182C8 )                                 exit

\ ******************************************************************************************

holinepu, 2018-09-09:
The decompiler program compiled by punyForth in RAM can be relocated in any address of flash rom now.
So, the application program can be compiled in RAM and then transfered to flash ROM for further use. 
\ lastword ` FFFF and ` 40260000  or  ` 3FFE88C0  !
` 4026DDCC ` 3FFE88C0                 !   \ only interpret? remains left in RAM for linking purpose
           ` 3FFE88C0   var-lastword  !   \ lastword change to interpret? where is in RAM
           ` 3FFE88E8   var-dp        !   \ here can be changed to allow more RAM spaces for further use
  see  see   see  words7
: t1 8  .  ;
: t2 88 .  ;
see t8

zeroflag, 2018-09-11:
@holinepu is the source code of this project available on github?

holinepu, 2018-09-12:
hi, Attila Magyar        Thank you for your interest in the source code of punyForth development system. Please bear in mind that the system is still under development so it looks not very clean. We need the help to promote the system so as most users will like it.  you can find the code here:https://groups.yahoo.com/neo/groups/armForth/files/esp8266%20punyForth/

 
 

   
 
|  | 不含病毒。www.avast.com  |

holinepu, 2018-09-19:
The program compiled by punyforth in RAM area can now be moved to any flash rom area by executing 
40280000  3FFE88C0>402xxxxx under esp8266 punyforth development system.  After flashing it into the chip we can execute words7 to see the org of each word. Likewise we can execute words6 to decompile the whole words.    
visit -->  https://groups.yahoo.com/neo/groups/armForth/files/esp8266%20punyForth/
words7

  ( lfa= ) 40218114 ( 00000000 )   ( cfa= 4021811C  <cfa>= 40239800 )  org t:+  dup
  ( lfa= ) 40218120 ( 40218114 )   ( cfa= 4021812C  <cfa>= 40239810 )  org t:+  drop
  ( lfa= ) 40218130 ( 40218120 )   ( cfa= 4021813C  <cfa>= 4023981C )  org t:+  swap
  ( lfa= ) 40218140 ( 40218130 )   ( cfa= 40218148  <cfa>= 40239830 )  org t:+  rot
  ( lfa= ) 4021814C ( 40218140 )   ( cfa= 40218158  <cfa>= 40239848 )  org t:+  2swap
  ( lfa= ) 4021815C ( 4021814C )   ( cfa= 40218168  <cfa>= 40239864 )  org t:+  2over
  ( lfa= ) 4021816C ( 4021815C )   ( cfa= 40218174  <cfa>= 4023987C )  org t:+  +
  ( lfa= ) 40218178 ( 4021816C )   ( cfa= 40218180  <cfa>= 40239890 )  org t:+  -
  ( lfa= ) 40218184 ( 40218178 )   ( cfa= 4021818C  <cfa>= 402398A4 )  org t:+  *
  ( lfa= ) 40218190 ( 40218184 )   ( cfa= 4021819C  <cfa>= 402398BC )  org t:+  /mod
  ( lfa= ) 402181A0 ( 40218190 )   ( cfa= 402181A8  <cfa>= 402398F8 )  org t:+  or
  ( lfa= ) 402181AC ( 402181A0 )   ( cfa= 402181B4  <cfa>= 4023990C )  org t:+  and
  ( lfa= ) 402181B8 ( 402181AC )   ( cfa= 402181C0  <cfa>= 40239920 )  org t:+  xor
  ( lfa= ) 402181C4 ( 402181B8 )   ( cfa= 402181D0  <cfa>= 40239934 )  org t:+  invert
  ( lfa= ) 402181D4 ( 402181C4 )   ( cfa= 402181E0  <cfa>= 40239948 )  org t:+  lshift
  ( lfa= ) 402181E4 ( 402181D4 )   ( cfa= 402181F0  <cfa>= 40239960 )  org t:+  rshift
  ( lfa= ) 402181F4 ( 402181E4 )   ( cfa= 40218200  <cfa>= 40239994 )  org t:+  _emit
  ( lfa= ) 40218204 ( 402181F4 )   ( cfa= 40218210  <cfa>= 402399C4 )  org t:+  abort
  ( lfa= ) 40218214 ( 40218204 )   ( cfa= 4021821C  <cfa>= 402399F0 )  org t:+  @
  ( lfa= ) 40218220 ( 40218214 )   ( cfa= 40218228  <cfa>= 40239A00 )  org t:+  c@
  ( lfa= ) 4021822C ( 40218220 )   ( cfa= 40218234  <cfa>= 40239A10 )  org t:+  !
  ( lfa= ) 40218238 ( 4021822C )   ( cfa= 40218240  <cfa>= 40239A24 )  org t:+  c!
  ( lfa= ) 40218244 ( 40218238 )   ( cfa= 4021824C  <cfa>= 40239A38 )  org t:+  [']
  ( lfa= ) 40218250 ( 40218244 )   ( cfa= 40218258  <cfa>= 40239A4C )  org t:+  <
  ( lfa= ) 4021825C ( 40218250 )   ( cfa= 40218268  <cfa>= 40239A64 )  org t:+  branch
  ( lfa= ) 4021826C ( 4021825C )   ( cfa= 40218278  <cfa>= 40239A74 )  org t:+  branch0
  ( lfa= ) 4021827C ( 4021826C )   ( cfa= 40218284  <cfa>= 40239A88 )  org t:+  >r
  ( lfa= ) 40218288 ( 4021827C )   ( cfa= 40218290  <cfa>= 40239A9C )  org t:+  r>
  ( lfa= ) 40218294 ( 40218288 )   ( cfa= 4021829C  <cfa>= 40239AB0 )  org t:+  i
  ( lfa= ) 402182A0 ( 40218294 )   ( cfa= 402182A8  <cfa>= 40239AC0 )  org t:+  j
  ( lfa= ) 402182AC ( 402182A0 )   ( cfa= 402182B8  <cfa>= 40239AD0 )  org t:+  execute
  ( lfa= ) 402182BC ( 402182AC )   ( cfa= 402182C8  <cfa>= 40239ADC )  org t:+  exit
  ( lfa= ) 402182CC ( 402182BC )   ( cfa= 402182D4  <cfa>= 40239AEC )  org t:+  sp@
  ( lfa= ) 402182D8 ( 402182CC )   ( cfa= 402182E0  <cfa>= 40239AFC )  org t:+  sp!
  ( lfa= ) 402182E4 ( 402182D8 )   ( cfa= 402182EC  <cfa>= 40239B0C )  org t:+  rp@
  ( lfa= ) 402182F0 ( 402182E4 )   ( cfa= 402182F8  <cfa>= 40239B1C )  org t:+  rp!
  ( lfa= ) 402182FC ( 402182F0 )   ( cfa= 4021830C  <cfa>= 40239B2C )  org t:+  readchar
  ( lfa= ) 40218360 ( 402182FC )   ( cfa= 4021836C  <cfa>= 40239B2C )  org t:+  xpause
  ( lfa= ) 4021837C ( 40218360 )   ( cfa= 40218390  <cfa>= 40239B58 )  org t:+  readchar-wait
  ( lfa= ) 40218394 ( 4021837C )   ( cfa= 402183A8  <cfa>= 40239B88 )  org t:+  readchar-nowait
  ( lfa= ) 402183AC ( 40218394 )   ( cfa= 402183B8  <cfa>= 40239BB8 )  org t:+  over
  ( lfa= ) 402183BC ( 402183AC )   ( cfa= 402183C8  <cfa>= 40239BC8 )  org t:+  -rot
  ( lfa= ) 402183CC ( 402183BC )   ( cfa= 402183D8  <cfa>= 40239BE0 )  org t:+  2dup
  ( lfa= ) 402183DC ( 402183CC )   ( cfa= 402183E8  <cfa>= 40239BF8 )  org t:+  2drop
  ( lfa= ) 402183EC ( 402183DC )   ( cfa= 402183F8  <cfa>= 40239C04 )  org t:+  4drop
  ( lfa= ) 402183FC ( 402183EC )   ( cfa= 40218408  <cfa>= 40239C10 )  org t:+  cells
  ( lfa= ) 4021840C ( 402183FC )   ( cfa= 40218414  <cfa>= 40239C28 )  org t:+  >
  ( lfa= ) 40218418 ( 4021840C )   ( cfa= 40218420  <cfa>= 40239C40 )  org t:+  =
  ( lfa= ) 40218424 ( 40218418 )   ( cfa= 4021842C  <cfa>= 40239C58 )  org t:+  <>
  ( lfa= ) 40218430 ( 40218424 )   ( cfa= 40218438  <cfa>= 40239C70 )  org t:+  <=
  ( lfa= ) 4021843C ( 40218430 )   ( cfa= 40218444  <cfa>= 40239C88 )  org t:+  >=
  ( lfa= ) 40218448 ( 4021843C )   ( cfa= 40218450  <cfa>= 40239CA0 )  org t:+  1+
  ( lfa= ) 40218454 ( 40218448 )   ( cfa= 4021845C  <cfa>= 40239CB0 )  org t:+  1-
  ( lfa= ) 40218460 ( 40218454 )   ( cfa= 40218468  <cfa>= 40239CC0 )  org t:+  0=
  ( lfa= ) 4021846C ( 40218460 )   ( cfa= 40218474  <cfa>= 40239CD4 )  org t:+  0<>
  ( lfa= ) 40218478 ( 4021846C )   ( cfa= 40218480  <cfa>= 40239CE8 )  org t:+  0<
  ( lfa= ) 40218484 ( 40218478 )   ( cfa= 4021848C  <cfa>= 40239CFC )  org t:+  0>
  ( lfa= ) 40218490 ( 40218484 )   ( cfa= 40218498  <cfa>= 40239D14 )  org t:+  ms@
  ( lfa= ) 4021849C ( 40218490 )   ( cfa= 402184A8  <cfa>= 40239D44 )  org t:+  _type
  ( lfa= ) 402184AC ( 4021849C )   ( cfa= 402184C0  <cfa>= 40239D74 )  org t:+  uart-set-bps
[... cut ...]
  ( 4026C0D4  4021824C )                                 ( ['] )    00000001
  ( 4026C0DC  4026B6DC )                                 word.#
  ( 4026C0E0  40268B50 )                                 +!
  ( 4026C0E4  40218268 )                                 branch     FFFFFF2C
  ( 4026C0EC  4026BF50 )                                 r>*
  ( 4026C0F0  4021812C )                                 drop
  ( 4026C0F4  4026BFB0 )                                 .freemem
  ( 4026C0F8  402182C8 )                                 exit

holinepu, 2018-09-28:
hi, 
    It seems like that no one cares about the cross-compiler that can be better to do the job. Ok! we'll change to the possibility to use punyForth itself to compile the source code and then transfer it to the console via rs232. That way we can obtain the hex file by executing , for example , 40280000 >flash. Now we can convert it to 40280000.bin file to be written to flash rom by flasher.exe. Wouldn't  that be great to finish the job!                                 Blees you by the name of Forth                           holi

[40260000.zip](https://github.com/zeroflag/punyforth/files/2426972/40260000.zip)

holinepu, 2018-09-29:
寄件者： chang luke <holinepu@yahoo.com.tw>收件者： Peter Forth <peter4th2017@gmail.com>寄件時間： 2018年9月29日 星期六 上午11:46:25 [GMT+8]主旨： Re: Punyforth and Win32forth , etc
  hi, Peter Forth and all the others       I have some experience in ASYST which use OVERLAY a lot. I still keep one at hand. Frankly speaking I didn't use it very well at that time. Maybe I'll recall that again.>  So the technical word, to define your work is you have created  an "OVERLAY capability to PunyForth" !
If it has enough RAM space then we don't have to have OVERLAY to switch between different applications. Still if we can get the OVERLAY capability that would be better. It is better to have than nothing, right?
The attached file temp.rar is for wifi and that sort of things. It works fine. I tried to compile ubernew163.forth into it. It was not very stable, so I switch to the other version ( older ? ) temp1.rar  and  ran dis-compiler in punyforth very successfully. I urge you guys to try this version out first to identify with our endeavor.  Ok!?                             Best  regard                                                 Be Forth with you                             holi  ( luke )

as for cross-compiler development system can still be used in cooperative with punyforth. under win32for  ( 4.2.67x )        fload  punyforth163.f      see  words    cls  words'    
see  /mod  \ *******************************************************************************  \  40218190  40218184  \  40218194  04 2F 6D 6F 64  \ -------------------------------------------------------------------------------                                   code  /mod  (  4021819C  402398BC          )  (  402398BC  38 0F 4B FF       )                  {drop>R3}  (  402398C0  28 0F 4B FF       )                  {drop>R2}  (  402398C4  12 C1 E4          )                     R1-1C  (  402398C7  09 61             )                  (R1+18)=R0  (  402398C9  89 41             )                  (R1+10)=R8  (  402398CB  D9 31             )                  (R1+0C)=RD  (  402398CD  E9 21             )                  (R1+8)=RE  (  402398CF  F2 61 01          )                  (R1+4)=RF  (  402398D2  01 F9 FF          )                  R0=40101168                 \ ( 402398B8 )= 40101168  (  402398D5  C0 00 00          )                  callx0_R0  (  402398D8  F8 11             )                  RF=(R1+4)  (  402398DA  E8 21             )                  RE=(R1+8)  (  402398DC  D8 31             )                  RD=(R1+0C)  (  402398DE  88 41             )                  R8=(R1+10)  (  402398E0  08 61             )                  R0=(R1+18)  (  402398E2  12 C1 1C          )                     R1+1C  (  402398E5  F2 CF FC 39 0F    )                  {dup<R3}  (  402398EA  F2 CF FC 29 0F    )                  {dup<R2}  (  402398EF  88 0E             )                  R8=(RE)  (  402398F1  4B EE             )                     RE+4  (  402398F3  98 08             )                  R9=(R8)  (  402398F5  A0 09 00          )                  jmp_R9
    在 2018年9月29日 星期六 上午5:40:13 [GMT+8]， Peter Forth<peter4th2017@gmail.com> 寫道：  
 
 Thanks, I'll check it out. 

When I used to program in LMI for DOS   those kind of files that were binaries pre-compiled, were
called    --  OVERLAYS  --.(other Forths also named this kind of binaries Overlays, and some other compilers
too, I think Turbo C and Turbo Pascal, had that possiblity too, I do not remember well)

So the technical word, to define your work is you have created  an "OVERLAY capability to PunyForth" !

We used this to store the EDITOR, the ASSEMBLER the Native code compiler, and any TOOL that
could be used for a task, and then when the task was complete, was not necessary anymore to have
it on RAM, this was very useful because RAM was limited to 64K. (The other secondary use was for distribution
of software  tools, without giving a third party access to your full source. )

Thanks again

Peter 

On Fri, Sep 28, 2018 at 12:50 PM chang luke <holinepu@yahoo.com.tw> wrote:

 hi,       The compiled code in RAM area can now be moved to flash ROM by executing   402688c0  >flash.  And after executing      forget   backref,   we can add new program into fresh RAM and later move to another address in flash ROM, say, 40260000  >flash.  We can link it to the previous one and so on.           good luck!                   holi
    在 2018年9月23日 星期日 下午5:01:08 [GMT+8]， chang luke<holinepu@yahoo.com.tw> 寫道：  
 
  Hi  Peter Forth !     It seems like that you didn't join the win32forth groups before. It's kind of pity that the Win32forth discussion groups lacking such kind of professional guy like you to join. I seldom join the discussion but still I can get the new information from there. I strongly urge you to subscribe win32forth discussion forum so as you can get the newly updated files from ARMForth groups by killing two birds with one stone. For now I just send you the newly updated file combined into one file to you so as you and the others don't have to find a way around.  And  here you have to go throw the following  procedures to expand the memory of the application memory.  Tell me if you've succeeded.
************************************************************************************************      before floading this file, we have to prepare the followings....\ with-img fsave Win32for    <-- modifying this line to the next line. ( in extend.f line 185 )  with-img fsave F.exe       \   thanks Cheng-Hong Chen  [... cut ...]
be done using Win32forth + VisualFOrth (making the screens and graphics) and
ESP8266 Punyforth.  Alltogether.  I think the Sky is the limit.
It is a good combination for teachers of technology school.

Anything you need I am at your disposal.

Sincerely 
Peter

( I am also setting up a web page with things I learn here and there, or things
I would like others  know and have a smoother start in Win32forth. Also  a 
book and PDF collection, and videos as you already know ).
https://sites.google.com/view/ win32forth/
  
 
|  | 不含病毒。www.avast.com  |

holinepu, 2018-10-04:
hi,  Gentlemen        I've been trying to make the code in RAM area to burn into flash ROM area. It is very successful now.  The main tricky program to do this is this:  :  modify(3FFE88C0~heap-400)  ( a -- n )     >r*     r@*      heap-start  ( ` 3ffe88a0 )  ` 20  +    heap-end  ` 400  -   between     r@*  @   heap-end   ` 400  -                    heap-end             between  not  and     r@*  @   ` 3ffe0000                             heap-start           between  not  and
     r@*  @   ` ffff0000   and  heap-end  ` ffff0000  and  ( ` 3ffe0000 ) =             and 
     if      r>*   @       402XXXXX-3FFE88C0   @   +   exit
     then    r>*   @    ; 
The attached file 40260000.forth is the main dis-compiler program. You can burn 40260000.forth and either one of the two .bin file to the flash ROM, 4026_171.bin / 4026_4026596c.bin.  Upon power on, we can execute words or  words7 to see the dictionary structure. Try to find the lfa of the last word in flash ROM   and executing it by:  `  4026b2f4   heap-start  32  +  !    Then we can execute the following words along to see what's going on.   words    forget  interpret?   words    see see                                                                                                                                                                                                                                 4026B2F0 org'  402182C8 |||| 4026B2C0 |||| 3230340C |||| 30303036 |||| \  ..!@..&@.4026000     4026B300 org'  6E652E30 |||| 00000064 |||| 40239B2C |||| 402182C8 |||| \  0.end...,.#@..!@     4026B310 org'  FFFFFFFF |||| FFFFFFFF |||| FFFFFFFF |||| FFFFFFFF |||| \  ................
We still have problems relating to wifi.  It needs your help. Best   regards                                                                                                        holi ( Luke )
    在 2018年9月29日 星期六 下午10:13:16 [GMT+8]， chang luke<holinepu@yahoo.com.tw> 寫道：  
 
  Owing to the fact that the compiled code, i.e. 40260000.bin has been written into flash rom, There is no need to write the source text file ( ubernew163.forth ) into the flash rom to be compiled after reset or power on.  Instead we can write ( burn ) the simpler source text file ( ram163.forth ) into the flash rom, so as to connect the lfa of the last word in flash with the lfa of first.program.in.ram.
由於編譯過的程式碼  40260000.bin 已燒入 flash ROM內   所以不需再燒入原始程式 ubernew163.forth    改燒入 ram163.forth 以連接 flash rom 最後一個的 lfa 到 first.program.in.ram 的 lfa                                              : first.program.in.ram ;1076289860  1073645760  !   \ it needs at least 3 lines of text    ` 4026E144     hex: ffff  and  hex: 40260000    or  heap-start  ` 20  +  !\ ( ` 4026E144 )   1076289860           heap-start  32  +  !\ ( ` 4026E144 )   lastword.in.flash           ` 3ffe88c0  !\       3ffe88c0 is lfa of first.program.in.ram == lastword.s     words     words7      see see  ......
    在 2018年9月29日 星期六 下午12:11:10 [GMT+8]， chang luke<holinepu@yahoo.com.tw> 寫道：  
 
  
 寄件者： chang luke <holinepu@yahoo.com.tw>收件者： Peter Forth <peter4th2017@gmail.com>寄件時間： 2018年9月29日 星期六 上午11:46:25 [GMT+8]主旨： Re: Punyforth and Win32forth , etc
  hi, Peter Forth and all the others       I have some experience in ASYST which use OVERLAY a lot. I still keep one at hand. Frankly speaking I didn't use it very well at that time. Maybe I'll recall that again.>  So the technical word, to define your work is you have created  an "OVERLAY capability to PunyForth" !
If it has enough RAM space then we don't have to have OVERLAY to switch between different applications. Still if we can get the OVERLAY capability that would be better. It is better to have than nothing, right?
The attached file temp.rar is for wifi and that sort of things. It works fine. I tried to compile ubernew163.forth into it. It was not very stable, so I switch to the other version ( older ? ) temp1.rar  and  ran dis-compiler in punyforth very successfully. I urge you guys to try this version out first to identify with our endeavor.  Ok!?                             Best  regard                                                 Be Forth with you                             holi  ( luke )

as for cross-compiler development system can still be used in cooperative with punyforth. under win32for  ( 4.2.67x )        fload  punyforth163.f      see  words    cls  words'    
see  /mod  \ *******************************************************************************  \  40218190  40218184  \  40218194  04 2F 6D 6F 64  \ -------------------------------------------------------------------------------                                   code  /mod  (  4021819C  402398BC          )  (  402398BC  38 0F 4B FF       )                  {drop>R3}  (  402398C0  28 0F 4B FF       )                  {drop>R2}  (  402398C4  12 C1 E4          )                     R1-1C  (  402398C7  09 61             )                  (R1+18)=R0  (  402398C9  89 41             )                  (R1+10)=R8  (  402398CB  D9 31             )                  (R1+0C)=RD  (  402398CD  E9 21             )                  (R1+8)=RE  (  402398CF  F2 61 01          )                  (R1+4)=RF  (  402398D2  01 F9 FF          )                  R0=40101168                 \ ( 402398B8 )= 40101168  (  402398D5  C0 00 00          )                  callx0_R0  (  402398D8  F8 11             )                  RF=(R1+4)  (  402398DA  E8 21             )                  RE=(R1+8)  (  402398DC  D8 31             )            [... cut ...]
be done using Win32forth + VisualFOrth (making the screens and graphics) and
ESP8266 Punyforth.  Alltogether.  I think the Sky is the limit.
It is a good combination for teachers of technology school.

Anything you need I am at your disposal.

Sincerely 
Peter

( I am also setting up a web page with things I learn here and there, or things
I would like others  know and have a smoother start in Win32forth. Also  a 
book and PDF collection, and videos as you already know ).
https://sites.google.com/view/ win32forth/
  
 
|  | 不含病毒。www.avast.com  |

holinepu, 2018-11-11:
hi,   I've completed the function of transfering the code compiled in ram area to the flash Rom area. It is quite easy to use it to extend the program now.  I've already send you the pf.rar file, see if you could try it out, please tell me if you succeed.                            Best regard                                holi

holinepu, 2019-01-25:
hi, everybody         The attached file is the complete source file for developing RISC-V RV32I, RV32M aiming to mecrispForth. Hope this does help to the progress of the development.                                                                 holi