zeroflag/punyforth #36: Need netcon-available function

CraigLindley, 2017-09-18:
In trying to do streaming with Punyforth I am using the netcon-read function. This is a blocking function that won't return until at least some bytes have been received. The blocking nature of this function means I cannot do other meaningful work in my code while waiting for bytes to be received. What I need is a function that I can call (like netcon-available) that will tell me how many bytes are available and if none are, I can execute other code until some data arrives.

Then I could use tasks: one for receiving data and putting it into a FIFO and another that would remove the data from the FIFO and process it.

zeroflag, 2017-09-18:
I can't find any function in http://lwip.wikia.com/wiki/Netconn_API that could be used for implementing netcon-available. Netcon is a sequential API by design. 

But I don't think you need this. You can still do things in a different task while your other task is waiting. There is a _pause_ in netcon-read that yields the control to other tasks while it's waiting for the data. So you can start one task that reads the socket and an other one that does something else. In the readme there are some examples and documentation about this.

An other option is to set a small read timeout on a netcon socket, catch the ERTIMEOUT exception do something else, and retry.
```forth
<timeout-in-seconds> <some-netcon-socket> netcon-read-timeout!
```

But I think the task based approach is better.

CraigLindley, 2017-09-18:
Thanks Attila, I'll give it a shot

-- 
Craig Lindley / Heather Hubbard

495's Recordings: craigandheather.net/495spage.html
New Recordings: craigandheather.net/cnmpage.html
Latest rock CD: craigandheather.net/oneinarow.html
Latest non-rock CD: craigandheather.net/craigdoesfingerstyle.html

Personal Website: craigandheather.net

Phone: (719) 495-1873
Cell: (719) 502-7925

If you’re one in a million, there are now seven thousand people exactly
like you.

closed 2018-06-20 by zeroflag