zeroflag/punyforth #37: when i flash puny forth i get the same error , see file

kedewey, 2018-02-23:
kendo@kdewey:~/Downloads/punyforth-master/arch/esp8266/bin > python flash.py /dev/ttyUSB0
Selected modules: ALL
Flashing binaries..
Flashing 0x0: rboot.bin, 0x1000: blank_config.bin, 0x2000: punyforth.bin
Connecting...
Erasing flash...
Took 0.16s to erase flash block
Writing at 0x00000000... (25 %)
Writing at 0x00000400... (50 %)
Writing at 0x00000800... (75 %)
Writing at 0x00000c00... (100 %)
Wrote 4096 bytes at 0x00000000 in 0.4 seconds (83.2 kbit/s)...
Erasing flash...
Took 0.17s to erase flash block
Writing at 0x00001000... (50 %)
Writing at 0x00001400... (100 %)
Wrote 2048 bytes at 0x00001000 in 0.2 seconds (83.7 kbit/s)...
Erasing flash...
Took 2.22s to erase flash block
Writing at 0x00002000... (0 %)
Writing at 0x00002400... (0 %)
Writing at 0x00002800... (0 %)
Writing at 0x00002c00... (1 %)
Writing at 0x00003000... (1 %)
Writing at 0x00003400... (1 %)
Writing at 0x00003800... (2 %)
Writing at 0x00003c00... (2 %)
Writing at 0x00004000... (2 %)
Writing at 0x00004400... (3 %)
Writing at 0x00004800... (3 %)
Writing at 0x00004c00... (3 %)
Writing at 0x00005000... (4 %)
Writing at 0x00005400... (4 %)
Writing at 0x00005800... (4 %)
Writing at 0x00005c00... (5 %)
Writing at 0x00006000... (5 %)
Writing at 0x00006400... (5 %)
Writing at 0x00006800... (6 %)
Writing at 0x00006c00... (6 %)
Writing at 0x00007000... (6 %)
Writing at 0x00007400... (7 %)
Writing at 0x00007800... (7 %)
Writing at 0x00007c00... (7 %)
Writing at 0x00008000... (8 %)
Writing at 0x00008400... (8 %)
Writing at 0x00008800... (8 %)
Writing at 0x00008c00... (9 %)
Writing at 0x00009000... (9 %)
Writing at 0x00009400... (9 %)
Writing at 0x00009800... (10 %)
Writing at 0x00009c00... (10 %)
Writing at 0x0000a000... (10 %)
Writing at 0x0000a400... (11 %)
Writing at 0x0000a800... (11 %)
Writing at 0x0000ac00... (11 %)
Writing at 0x0000b000... (12 %)
Writing at 0x0000b400... (12 %)
Writing at 0x0000b800... (12 %)
Writing at 0x0000bc00... (13 %)
Writing at 0x0000c000... (13 %)
Writing at 0x0000c400... (13 %)
Writing at 0x0000c800... (14 %)
Writing at 0x0000cc00... (14 %)
Writing at 0x0000d000... (14 %)
Writing at 0x0000d400... (15 %)
Writing at 0x0000d800... (15 %)
Writing at 0x0000dc00... (15 %)
Writing at 0x0000e000... (16 %)
Writing at 0x0000e400... (16 %)
Writing at 0x0000e800... (16 %)
[... 312 lines cut ...]
Writing at 0x0005b400... (40 %)
Writing at 0x0005b800... (60 %)
Writing at 0x0005bc00... (80 %)
Writing at 0x0005c000... (100 %)
Wrote 5120 bytes at 0x0005b000 in 0.5 seconds (84.9 kbit/s)...
Erasing flash...
Took 0.13s to erase flash block
Writing at 0x0005d000... (50 %)
Writing at 0x0005d400... (100 %)
Wrote 2048 bytes at 0x0005d000 in 0.2 seconds (85.4 kbit/s)...
Erasing flash...
Took 0.09s to erase flash block
Writing at 0x0005e000... (50 %)
Writing at 0x0005e400... (100 %)
Wrote 2048 bytes at 0x0005e000 in 0.2 seconds (85.4 kbit/s)...
Erasing flash...
Took 0.12s to erase flash block
Writing at 0x0005f000... (100 %)
Wrote 1024 bytes at 0x0005f000 in 0.1 seconds (85.4 kbit/s)...
Erasing flash...
Took 0.09s to erase flash block
Writing at 0x00060000... (16 %)

A fatal error occurred: Invalid head of packet
ken

zeroflag, 2018-02-23:
Why kind of board do you have? Can you try it with _--flashmode dio_ parameter?

zeroflag, 2018-05-12:
I added a note about this to the readme.

https://github.com/zeroflag/punyforth/commit/653ef3c05c2efe635a079647326b0d8b3e9d25dd#diff-04c6e90faac2675aa89e2176d2eec7d8

closed 2018-05-12 by zeroflag