zeroflag/punyforth #38: Problem with flashing modules

craiglindley, 2018-05-09:
I followed the documentation in the readme file and successfully flashed a esp8266 using the following commands on my mac:
python flash.py /dev/tty.SLAB_USBtoUART

My understanding is that without specifying which modules to program, all modules would be programmed including the GPIO module. Unfortunately when I type help to see all defined words, words like GPIO_HIGH or GPIO_IN, etc. are not found. Further more when I try to examine the value of GPIO_HIGH it is undefined.

Then I attempted to specify the GPIO module directly using the following command line it still doesn't work.
python flash.py /dev/tty.SLAB_USBtoUART --modules core gpio flash

What am I doing wrong?

I should also say that the docs still shows the use of modules.py file to control which forth modules to be included in uber.forth. It now seems uber.forth is no longer used and the modules.py functionality is now included in the flash.py file. Please update the docs to match the code when you get a chance.

zeroflag, 2018-05-09:
Yeah, in the recent version this changed quite much. The flash.py flashes everything by default, so GPIO is there. In order to use it you have to load it by either typing

```
(stack) GPIO load
```

in the REPL, or by adding _GPIO load_ to the beginning of your program (or right before you wan't to use GPIO related words). This is similar than an import in other languages. If you load something multiple times, it will eventually eat up the dictionary space so make sure you load it once.

So normally you don't need to worry about what modules you want to have up front. You just flash everything and later you can load anything into memory if needed.

I still need to update the doc about this.

craiglindley, 2018-05-09:
So I would have to "FLASH" load for example to use that module? And the
same would be true for any of the other modules listed in AVAILABLE_MODULES
in flash.py?

So all modules are loaded into flash but only are moved to RAM when they
are loaded?

-- 
Craig Lindley / Heather Hubbard

495's Recordings: craigandheather.net/495spage.html
New Recordings: craigandheather.net/cnmpage.html
Latest rock CD: craigandheather.net/oneinarow.html
Latest non-rock CD: craigandheather.net/craigdoesfingerstyle.html

Personal Website: craigandheather.net

Phone: (719) 495-1873
Cell: (719) 502-7925

If you’re one in a million, there are now seven thousand people exactly
like you.