zeroflag/punyforth #39: Unexplained Crash

craiglindley, 2018-05-10:
I am currently doing some graphics programming on the esp8266 and have run into a crash that I cannot explain. When I execute the following code and let it run for a while, it always crashes.

128 constant: displayWidth
160 constant: displayHeight

displayHeight 2 / constant: MIDY
displayWidth  2 / constant: MIDX

\ Manhattan distance from middle of screen
: dist 
    MIDY - abs swap MIDX - abs +
;

: test

  displayWidth 0
  do
     displayHeight 0
     do
        j i dist . cr
     loop
  loop
;

Can anyone see anything wrong?

zeroflag, 2018-05-10:
I don't think anything wrong with the code at first glance, and it ran fine on a raspberry. Can you check the available freemem by typing in _freemem_?

craiglindley, 2018-05-10:
After I compile the code freemem say 17132. It takes running for about 30
seconds to see the issue.

-- 
Craig Lindley / Heather Hubbard

495's Recordings: craigandheather.net/495spage.html
New Recordings: craigandheather.net/cnmpage.html
Latest rock CD: craigandheather.net/oneinarow.html
Latest non-rock CD: craigandheather.net/craigdoesfingerstyle.html

Personal Website: craigandheather.net

Phone: (719) 495-1873
Cell: (719) 502-7925

If you’re one in a million, there are now seven thousand people exactly
like you.

craiglindley, 2018-05-10:
It must have something to do with writing to the serial port because if I replace the ".  cr" with a "drop" the code will run to completion.

craiglindley, 2018-05-10:
> Interesting. Was there any other hardware connected to some of the GPIOs
> of the esp? Tomorrow I'll try to run it on my board.

I have an LCD display connected but since I didn't load GPIO or the driver
it should not be an issue. I was thinking that my ESP8266 Puny Forth Loader
might be the issue but if I run the real code and not this pared down
version I get an exception 9 even when I am not sending output to the
serial port.

-- 
Craig Lindley / Heather Hubbard

495's Recordings: craigandheather.net/495spage.html
New Recordings: craigandheather.net/cnmpage.html
Latest rock CD: craigandheather.net/oneinarow.html
Latest non-rock CD: craigandheather.net/craigdoesfingerstyle.html

Personal Website: craigandheather.net

Phone: (719) 495-1873
Cell: (719) 502-7925

If you’re one in a million, there are now seven thousand people exactly
like you.

zeroflag, 2018-05-12:
I ran it on a real board and I haven't seen any problem. Have you tried running it without the display? Maybe something is wrong with the pin layout, do you use any of the RX/TX pins?

closed 2018-05-12 by zeroflag