zeroflag/punyforth #41: Punyforth not fully booting?

cefn, 2018-06-03:
May have spoken too soon in closing https://github.com/zeroflag/punyforth/issues/40 

After replacing with the latest esptool.py I thought the flashing process had completed but there still seems to be something wrong with the punyforth session which comes up after the flashing reports success. I tried resorting to a NodeMCUv2 device which has a CP2102 and ESP8266 with 4MB but this behaves identically.

I connected via `screen /dev/ttyUSB0 115200` and it shows `(stack)` but the terminal session is non-responsive - none of the keys you press appear as echoed characters and typing the example test command `println: "hello" ` does nothing.

On closer inspection of the example terminal session you show on the main page, I should probably be seeing the phrase "Punyforth ready" which never arrives, and not just (stack).

If I stay connected to /dev/ttyUSB0 and press reset it shows the following after the boot messages...

```
pp_task_hdl : 3fff5c20, prio:14, stack:512
pm_task_hdl : 3fff5680, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4950, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jan 10 2018 17:21:17
phy ver: 273, pp ver: 8.3

Loading Punyforth
mode : softAP(5e:cf:7f:1e:13:e0)
add if1
bcn 100
.......
(stack)
```

Below is a log of the upload process in case there's anything which might help...

```
$ python2 flash.py /dev/ttyUSB1
Selected modules: ALL
Flashing binaries..
Flashing 0x0: rboot.bin, 0x1000: blank_config.bin, 0x2000: punyforth.bin
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.4.0-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0040
Compressed 3104 bytes to 2169...
Wrote 3104 bytes (2169 compressed) at 0x00000000 in 0.2 seconds (effective 126.7 kbit/s)...
Hash of data verified.
Compressed 2048 bytes to 23...
Wrote 2048 bytes (23 compressed) at 0x00001000 in 0.0 seconds (effective 2326.9 kbit/s)...
Hash of data verified.
Compressed 310800 bytes to 203795...
Wrote 310800 bytes (203795 compressed) at 0x00002000 in 18.0 seconds (effective 137.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Flashing LAYOUT.tmp
esptool.py v2.4.0-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 392 bytes to 184...
Wrote 392 bytes (184 compressed) at 0x00051000 in 0.0 seconds (effective 149.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Flashing 0x52000: APP.tmp, 0x53000: CORE.tmp, 0x56000: DHT22.tmp, 0x57000: FLASH.tmp, 0x58000: FONT57.tmp, 0x59000: GPIO.tmp, 0x5a000: MAILBOX.tmp, 0x5b000: NETCON.tmp, 0x5d000: NTP.tmp, 0x5e000: PING.tmp, 0x5f000: SONOFF.tmp, 0x60000: SSD1306I2C.tmp, 0x62000: SSD1306SPI.tmp, 0x64000: TASKS.tmp, 0x65000: TCPREPL.tmp, 0x66000: TURNKEY.tmp, 0x67000: WIFI.tmp, 0x68000: EVENT.tmp, 0x69000: RINGBUF.tmp, 0x6a000: DECOMP.tmp, 0x6b000: PUNIT.tmp
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.4.0-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 40 bytes to 39...
Wrote 40 bytes (39 compressed) at 0x00052000 in 0.0 seconds (effective 39.8 kbit/s)...
Hash of data verified.
Compressed 11044 bytes to 3502...
Wrote 11044 bytes (3502 compressed) at 0x00053000 in 0.3 seconds (effective 282.2 kbit/s)...
Hash of data verified.
Compressed 1768 bytes to 722...
Wrote 1768 bytes (722 compressed) at 0x00056000 in 0.1 seconds (effective 208.0 kbit/s)...
Hash of data verified.
Compressed 1528 bytes to 714...
Wrote 1528 bytes (714 compressed) at 0x00057000 in 0.1 seconds (effective 179.7 kbit/s)...
Hash of data verified.
Compressed 3864 bytes to 1720...
Wrote 3864 bytes (1720 compressed) at 0x00058000 in 0.2 seconds (effective 198.2 kbit/s)...
Hash of data verified.
Compressed 800 bytes to 361...
Wrote 800 bytes (361 compressed) at 0x00059000 in 0.0 seconds (effective 177.6 kbit/s)...
Hash of data verified.
Compressed 340 bytes to 172...
Wrote 340 bytes (172 compressed) at 0x0005a000 in 0.0 seconds (effective 136.1 kbit/s)...
Hash of data verified.
Compressed 4252 bytes to 1329...
Wrote 4252 bytes (1329 compressed) at 0x0005b000 in 0.1 seconds (effective 278.7 kbit/s)...
Hash of data verified.
Compressed 1080 bytes to 542...
Wrote 1080 bytes (542 compressed) at 0x0005d000 in 0.1 seconds (effective 163.1 kbit/s)...
Hash of data verified.
Compressed 1096 bytes to 579...
Wrote 1096 bytes (579 compressed) at 0x0005e000 in 0.1 seconds (effective 156.8 kbit/s)...
Hash of data verified.
Compressed 616 bytes to 307...
Wrote 616 bytes (307 compressed) at 0x0005f000 in 0.0 seconds (effective 153.6 kbit/s)...
Hash of data verified.
Compressed 5224 bytes to 1861...
Wrote 5224 bytes (1861 compressed) at 0x00060000 in 0.2 seconds (effective 248.7 kbit/s)...
Hash of data verified.
Compressed 5276 bytes to 1751...
Wrote 5276 bytes (1751 compressed) at 0x00062000 in 0.2 seconds (effective 263.9 kbit/s)...
Hash of data verified.
Compressed 3260 bytes to 1083...
Wrote 3260 bytes (1083 compressed) at 0x00064000 in 0.1 seconds (effective 258.1 kbit/s)...
Hash of data verified.
Compressed 1760 bytes to 714...
Wrote 1760 bytes (714 compressed) at 0x00065000 in 0.1 seconds (effective 206.6 kbit/s)...
Hash of data verified.
Compressed 804 bytes to 416...
Wrote 804 bytes (416 compressed) at 0x00066000 in 0.0 seconds (effective 156.9 kbit/s)...
Hash of data verified.
Compressed 1444 bytes to 600...
Wrote 1444 bytes (600 compressed) at 0x00067000 in 0.1 seconds (effective 199.6 kbit/s)...
Hash of data verified.
Compressed 328 bytes to 193...
Wrote 328 bytes (193 compressed) at 0x00068000 in 0.0 seconds (effective 119.6 kbit/s)...
Hash of data verified.
Compressed 1392 bytes to 446...
Wrote 1392 bytes (446 compressed) at 0x00069000 in 0.0 seconds (effective 252.9 kbit/s)...
Hash of data verified.
Compressed 3312 bytes to 1257...
Wrote 3312 bytes (1257 compressed) at 0x0006a000 in 0.1 seconds (effective 230.1 kbit/s)...
Hash of data verified.
Compressed 1972 bytes to 750...
Wrote 1972 bytes (750 compressed) at 0x0006b000 in 0.1 seconds (effective 221.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

kbr2000, 2018-06-03:
Hey there, i'm experiencing the same problem (using an ESP-01 module with a recent punyforth clone from yesterday). 

The flashing procedure works, so does booting, but it doesn't print the "Punyforth ready." message, and seems to hang after displaying "(stack)". 

I'm using minicom instead at 115200 baud, 8N1, through a PL2303 based USB-to-serial adapter. 

The ESP seems to function otherwise, as i can see a WiFi AP appearing with a SSID corresponding with my module's MAC address. 

My next step will likely be retesting the PL2303 (which is working bidirectionally since i was able to both flash and receive the boot messages), and observing the serial communication closer to see if i'm missing something. 

See you all back here!

zeroflag, 2018-06-03:
Hi, can you try this out with coolterm, using the settings from the readme.md? I remember having some problems with screen. Probably it never worked with screen or screen should be configured differently.

GerardSontag, 2018-06-03:
Hello, i had similar problem with TeraTerm/punyforth neither with TeraTerm/other kind of forth.
If i remember correctly punyforth is not doing a reset/init on the UART port before using it and i suspect this is the reason why some of us get troubles.

zeroflag, 2018-06-03:
> If i remember correctly punyforth is not doing a reset/init on the UART port before using 

Thanks for the tip @GerardSontag, I'll investigate this.

cefn, 2018-06-03:
Switched to Windows (Virtualbox based on Free Windows 10 90-day 'Edge' tester from Microsoft). That's because Coolterm is only available as a 32-bit application on Linux, and it was looking like a world of pain to install multiarch and a whole set of i386 desktop libs just for Coolterm.

Coolterm does indeed function against the flashed punyforth on NodeMCU-M (ESP8285) following my flashing with `--flashmode dout`

I was then able to prove that [miniterm](http://pyserial.readthedocs.io/en/latest/tools.html#miniterm) could also connect and behave as expected if I avoided my usual --raw flag (which disables line mode), and added a new --echo flag. Connecting like this was successful from CMD.exe after installing Python3 into windows (adding it to the path). For Pip instructions try... https://vgkits.org/blog/pip3-howto/

```
pip3 install pyserial
python -m serial.tools.miniterm --echo COM3 115200
```

cefn, 2018-06-03:
And now also proven to work back on Linux...

To install miniterm (after [installing pip3](https://vgkits.org/blog/pip3-howto/))...

```
pip3 install pyserial
```

Then run it like...

```
python3 -m serial.tools.miniterm --echo /dev/ttyUSB1 115200
```

closed 2018-06-03 by cefn