zeroflag/punyforth #44: Unexplained crash on ESP8266

Thoric314, 2019-02-12:
It looks like I have the same problem as what was demonstrated in the issue of 10 May 2018.

As it is closed, I reopen here (the code is also not the same : here I take it from Readme.md)

I use the punyforth version taken from github (ESP8266 : nodemcu v0.9 is my device) and then also a version that I compiled by myself. Both presented the same stange behavior (driving to crash).

then:
(stack) **`5 0 do i . loop`**

Word has no interpretation semantics
0
Word has no interpretation semantics

(stack 5 0)

then:

(stack 5 0) **`10 0 do i . 2 +loop`**

Word has no interpretation semantics
0
Word has no interpretation semantics
Fatal exception (29): 
epc1=0x4023da00
epc2=0x00000000
epc3=0x4020ff20
excvaddr=0x00000000
depc=0x00000000
excsave1=0x40215a90
Registers:
a0 40215a90 a1 3fffbfd0 a2  00000000 a3  3fff7870
a4  00000001 a5  00000000 a6  4021ddc6 a7  00000000
a8  00000000 a9  3ffea9b0 a10 60000000 a11 0000000a
a12 00000000 a13 3ffeec48 SAR 0000001e

Stack: SP=0x3fffbfd0
0x3fffbfd0: 00000000 00000000 00000000 00000000
0x3fffbfe0: 00000000 00000000 00000000 a5a5a5a5
0x3fffbff0: fab22c8a 0000014c 3fffbf80 79fa9aaa

Free Heap: 18228
_heap_start 0x3fff7a68 brk 0x3fffc85c supervisor sp 0x40000000 sp-brk 14244 bytes
arena (total_size) 19956 fordblks (free_size) 3984 uordblocks (used_size) 15972

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

I had to remove power and to connect again to see the prompt again!
Any idea is welcome but I think there is something wrong in the code or in my ESP8266.

zeroflag, 2019-02-12:
Hi @Thoric314,

This is an expected behaviour. _do_ + _loop_ are compile time only words, you can only use them inside a colon definition. In interpretation mode they're skipped and a warning message is printed out. But the input buffer is not dropped so further words are interpreted. Normally if you use the word _i_ inside a do loop you'll access to the loop variable which is stored on the return stack. But since _do_ and _loop_ are skipped because you're in interpretation mode and _i_ still consumes the return stack which messes up the outer interpreter.

To make this work you should use the loop inside a colon def

```forth
(stack) : whatever 5 0 do i . loop ; whatever
```

Maybe dropping the input buffer when a compile time word is used in interpretation mode would be a good idea. Let me think about it.

Attila

Thoric314, 2019-02-12:
Ooops, sorry for the noise!
So my suggestion would be to change the example in the Readme.md, perhaps
just putting it as you mentionned to me:

(stack) : whatever 5 0 do i . loop ; whatever

I was too fast!

Thanks again for the quick answer

Best regards!

Pat

Thoric314, 2019-02-13:
It works as expected! RTFM again 
Thanks

closed 2019-02-13 by Thoric314