zeroflag/punyforth #45: Memory troubles solved

Jos-Ven, 2019-02-15:
Hi,
I had some trouble to flash punyforth. When using:
python flash.py /dev/ttyUSB0 --main blinker.forth

I got:
Writing at 0x0005f000... (100 %)
Wrote 1024 bytes at 0x0005f000 in 0.1 seconds (84.5 kbit/s)...
Erasing flash...

A fatal error occurred: Invalid head of packet.

Then after reading some more stuff I entered:
python esptool.py  erase_flash
Reset the ESP8266
python flash.py /dev/ttyUSB0 --flashmode dio --main blinker.forth --modules gpio.forth
Ran it. And then:
python flash.py /dev/ttyUSB0 --flashmode dio --main blinker.forth --binary false

Now I got:
Writing at 0x0006b400... (100 %)
Wrote 2048 bytes at 0x0006b000 in 0.2 seconds (82.8 kbit/s)...

Leaving...

Finally, no memory error anymore. I think something was blocking access to the memory.

What is the highest possible address to which I can write to on my ESP8266 ESP-12F ?

Jos

zeroflag, 2019-02-18:
@Jos-Ven probably the problem was that your esp doesn't support Quad I/O mode which is used by default by flash.py. When you switched to dio it looks like it started working.

Jos-Ven, 2019-02-18:
Hi, 

Thank you, I think you are right.
Being a complete newbie for PunyForth I would like you to ask if there is
a quick way to get to the (stack) prompt without restarting the whole Forth-system
as abort does.

Jos

zeroflag, 2019-02-18:
@Jos-Ven, 

If you start Punyforth on the esp you should get the (stack) prompt automatically after the system booted. If you start some code that runs indefinitely in single task you can't access to the prompt until that code returns. However if you execute that code as a task in multitasking mode then you'll still see the prompt and you'll be able to interact with Punyforth further.

For example:

```forth

\ create a task for the consumer
0 task: mytask

: some-infinite-loop ( task -- )
  activate
  begin 
    println: "Still running..."
    1000 ms
    pause
  again ;

multi \ switch to multitask mode
mytask some-infinite-loop \ this will run the task in the background while the prompt is still active
```