zeroflag/punyforth #47: Complete word list with usage

hinsech475, 2019-02-18:
Hi to your team. I enjoy using Forth as a hobby on my Wemos D1. But PunyForth is giving me a hard time because I cannot find a complete list of all words and their usage. The word help is the only way I found to see all defined words, but without a description of the usage. Could you please tell me where I could find  documentation describing all PunyForth words and their usage. I appreciate your collaboration.

zeroflag, 2019-02-18:
Hi @hinsech475,

Since stack effect and other comments are not stored in the dictionary there is no way to access them during runtime.

You can use the word _help_ to see all available words in the dictionary and use the decompiler to see the decompiled source code of a given word.

```forth
DECOMP load \ first load the decompiler

(stack) decompile: \
16r00021C30:  <entercol>
16r00021C34:  key        <-                                      
16r00021C38:  crlf?        |                                     
16r00021C3C:  branch0    ->                                      
16r00021C40:  -12                                                
16r00021C44:  <exit>                                             
(stack)
```

This decompiles the word \\. This might help but if you're interested in the real source code and stack effects comments the only way to access those is to find the word in the github repository.

hinsech475, 2019-02-19:
Hi Attila.

For anyone porficient with Forth, the help and decompile  words are probably sufficient to understand the usage of the PunyForth vocabulary.

For a beginner, it is a challenge, but it will also teach him a lot if he persists.

I am a retiree and will probably take the time to extract a list of words, stack effects and functions from the assembly and Forth source codes. I have the time to do it but most people would not be that persistent.

A PDF file listing all words , their stack effect and their function would make learning PunyForth easier for beginners and intermediate users and would make it more accessible to most users.

At the start of a project, it is a lot easier to know which tools you have to do it and what they do, rather than figuring out on the way if you have the tool. But Forth allow you to crate it if its missing... if possible without re-inventing the wheel. This is where the vocabulary list would be useful.

You have done an excellent job with PunyForth, but it is a bit discouraging to figure out how to use it, like most Forth version. Anyway, it is a lot more interesting than assembling a puzzle.

I will now keep on playing with my Wemos D1 and PunyForth, Lua and  microPython...

Thanks for this excellent Forth version.

Christian Hinse