zeroflag/punyforth #52: highbit

Jos-Ven, 2019-08-22:
Hi,
At I am still quite happy with your Punyforth. Great work!
There is just 1 question for a primitive floating-point pack.

When highbit is defined as:
-1 1 rshift invert constant: highbit  
than highbit returns 214748364 
Other 32 bits systems return 2147483648
How should highbit be defined?
Thanks in advance.

Despite that problem the pack seem to work as you can see:

(stack) 10 s>f 3 s>f f/ f.
333333333e-8(stack) 
10 s>f 6 s>f f/ f.
1666666666e-9(stack)

Kind regards, Jos

klapauciusisgreat, 2020-05-24:
I'm curious which floating point packs are freely available. Jos, can you point so something interesting ?

I'd like to get a at least double precision library going for a project doing some astronomical toy (say, point antenna at moon) and most code for that does not really work with single precision.

Also, am I right at suspecting that a 'proper' floating point implementation requires to change the interpreter to recognize FP numbers ? It seems feasible but seems like a fairly decent chunk of work, especially when FP speed is not of the essence (e.g. Moon angular speed relative to earth is not that fast).

Jos-Ven, 2020-05-25:
Hi,

In the end I managed to port a simple high level floating point pack to 
Punyforth.
There is no floating-point stack. It uses the normal stack.
2147483647 1+ is used as the highbit.
Not sure that is right for Punyforth.
The pack needs to be tested.

Interested?
Jos

From: klapauciusisgreat
Sent: Monday, May 25, 2020 12:00 AM
To: zeroflag/punyforth
Cc: Jos ; Author
Subject: Re: [zeroflag/punyforth] highbit (#52)

I'm curious which floating point packs are freely available. Jos, can you 
point so something interesting ?

I'd like to get a at least double precision library going for a project 
doing some astronomical toy (say, point antenna at moon) and most code for 
that does not really work with single precision.

Also, am I right at suspecting that a 'proper' floating point implementation 
requires to change the interpreter to recognize FP numbers ? It seems 
feasible but seems like a fairly decent chunk of work, especially when FP 
speed is not of the essence (e.g. Moon angular speed relative to earth is 
not that fast).

klapauciusisgreat, 2020-05-25:
Yes, interested, as I didn't really find a high level FP library (I haven't
anything fort in a loong time).

Many thanks!

Max

Jos-Ven, 2020-05-26:
Hi,

See zenfloat.zip at the end of https://sites.google.com/site/win324th/smart-home
for a download.

Jos

From: Klapaucius 
Sent: Monday, May 25, 2020 5:54 PM
To: zeroflag/punyforth 
Cc: Jos ; Author 
Subject: Re: [zeroflag/punyforth] highbit (#52)

Yes, interested, as I didn't really find a high level FP library (I haven't
anything fort in a loong time).

Many thanks!

Max

klapauciusisgreat, 2020-05-27:
Very cool, thanks! It's always amazing how much you can do with so little
code in forth (though APL seems equally compact).

This won't be enough for me since for pointing at the moon I need a bunch
of trig functions. I'll consider adding floating point to punyforth
natively, but I first need to get it to work on esp32 devices at all. The
code is a bit gnarly - both on the esp-idf sdk and punyforth side, but it
seems doable.

I also found some  hints on how to make micropython double precision, I
will try that as well. It seems like a simpler way forward, and most esp32
devices have no problem running substantial python programs.

Anyway, thanks again (also for the winforth pages -- nice!)

Max

Jos-Ven, 2020-05-28:
Hi,

Please inform us when punyforth is working on a esp32.

Jos

From: Klapaucius
Sent: Wednesday, May 27, 2020 6:20 PM
To: zeroflag/punyforth
Cc: Jos ; Author
Subject: Re: [zeroflag/punyforth] highbit (#52)

Very cool, thanks! It's always amazing how much you can do with so little
code in forth (though APL seems equally compact).

This won't be enough for me since for pointing at the moon I need a bunch
of trig functions. I'll consider adding floating point to punyforth
natively, but I first need to get it to work on esp32 devices at all. The
code is a bit gnarly - both on the esp-idf sdk and punyforth side, but it
seems doable.

I also found some hints on how to make micropython double precision, I
will try that as well. It seems like a simpler way forward, and most esp32
devices have no problem running substantial python programs.

Anyway, thanks again (also for the winforth pages -- nice!)

Max