zeroflag/punyforth #58: Flash starts but doesn't finish properly

MayamaTakeshi, 2020-10-03:
Hi,
I'm trying to flash punyforth into my ESP-01 from my Ubuntu 18.04.
It seems to start fine but it suddenly ends without an error:

```
takeshi@takeshi-Prime-Series:~/extra/src/git/punyforth/arch/esp8266/bin$ python flash.py --flashmode dio /dev/ttyUSB0
Selected modules: ALL
Flashing binaries..
Flashing 0x0: rboot.bin, 0x1000: blank_config.bin, 0x2000: punyforth.bin
Connecting...
Erasing flash...
Took 0.16s to erase flash block
Writing at 0x00000000... (25 %)
Writing at 0x00000400... (50 %)
Writing at 0x00000800... (75 %)
Writing at 0x00000c00... (100 %)
Wrote 4096 bytes at 0x00000000 in 0.4 seconds (82.1 kbit/s)...
Erasing flash...
Took 0.16s to erase flash block
Writing at 0x00001000... (50 %)
Writing at 0x00001400... (100 %)
Wrote 2048 bytes at 0x00001000 in 0.2 seconds (82.8 kbit/s)...
Erasing flash...
Took 2.32s to erase flash block
Writing at 0x00002000... (0 %)
Writing at 0x00002400... (0 %)
Writing at 0x00002800... (0 %)
Writing at 0x00002c00... (1 %)
Writing at 0x00003000... (1 %)
Writing at 0x00003400... (1 %)
Writing at 0x00003800... (2 %)
Writing at 0x00003c00... (2 %)
Writing at 0x00004000... (2 %)
Writing at 0x00004400... (3 %)
Writing at 0x00004800... (3 %)
Writing at 0x00004c00... (3 %)
Writing at 0x00005000... (4 %)
Writing at 0x00005400... (4 %)
Writing at 0x00005800... (4 %)
Writing at 0x00005c00... (5 %)
Writing at 0x00006000... (5 %)
Writing at 0x00006400... (5 %)
Writing at 0x00006800... (6 %)
Writing at 0x00006c00... (6 %)
Writing at 0x00007000... (6 %)
Writing at 0x00007400... (7 %)
Writing at 0x00007800... (7 %)
Writing at 0x00007c00... (7 %)
Writing at 0x00008000... (8 %)
Writing at 0x00008400... (8 %)
Writing at 0x00008800... (8 %)
Writing at 0x00008c00... (9 %)
Writing at 0x00009000... (9 %)
Writing at 0x00009400... (9 %)
Writing at 0x00009800... (10 %)
Writing at 0x00009c00... (10 %)
Writing at 0x0000a000... (10 %)
Writing at 0x0000a400... (11 %)
Writing at 0x0000a800... (11 %)
Writing at 0x0000ac00... (11 %)
Writing at 0x0000b000... (12 %)
Writing at 0x0000b400... (12 %)
Writing at 0x0000b800... (12 %)
Writing at 0x0000bc00... (13 %)
Writing at 0x0000c000... (13 %)
Writing at 0x0000c400... (13 %)
Writing at 0x0000c800... (14 %)
Writing at 0x0000cc00... (14 %)
Writing at 0x0000d000... (14 %)
Writing at 0x0000d400... (15 %)
[... 384 lines cut ...]
Took 0.13s to erase flash block
Writing at 0x0006a000... (25 %)
Writing at 0x0006a400... (50 %)
Writing at 0x0006a800... (75 %)
Writing at 0x0006ac00... (100 %)
Wrote 4096 bytes at 0x0006a000 in 0.4 seconds (83.0 kbit/s)...
Erasing flash...
Took 0.13s to erase flash block
Writing at 0x0006b000... (50 %)
Writing at 0x0006b400... (100 %)
Wrote 2048 bytes at 0x0006b000 in 0.2 seconds (82.8 kbit/s)...

Leaving...
takeshi@takeshi-Prime-Series:~/extra/src/git/punyforth/arch/esp8266/bin$ echo $?
0
```
Based on output shown in other issues, i believe I should see:
```
Leaving...
Hard resetting via RTS pin...
```
But this didn't happen.
Anyway, after the above, when I reset the board, I can connect to it using a terminal emulator (tio) but nothing shows up, and nothing that I input gets echoed back.

Does anybody have any tips for me?

Jos-Ven, 2020-10-03:
Hi, 

The last time I used flash.py I used it in combination with picocom:
For picocom I use:
$ picocom /dev/ttyUSB0 -b 115200 --omap crlf --echo --imap delbs

In the past I also got some problems.
That was solved after erasing the flash memory see:
https://github.com/zeroflag/punyforth/issues/45
( try adding --main blinker.forth )
Not sure if it helps.

Jos

MayamaTakeshi, 2020-10-04:
Thanks.
The erase_flash command helped.

I experimented changing the flashmode: both qio and qout fail. 
Only dio terminates without errors but without Serial communication.
But with dout, although there are errors:
```
takeshi@takeshi-brastel:bin$ python esptool.py erase_flash
Connecting...
Erasing flash (this may take a while)...
takeshi@takeshi-brastel:bin$ python flash.py /dev/ttyUSB0 --flashmode dout
Selected modules: ALL
Flashing binaries..
Flashing 0x0: rboot.bin, 0x1000: blank_config.bin, 0x2000: punyforth.bin
Connecting...
Erasing flash...
Took 0.15s to erase flash block
Writing at 0x00000000... (25 %)
Writing at 0x00000400... (50 %)
Writing at 0x00000800... (75 %)
Writing at 0x00000c00... (100 %)
Wrote 4096 bytes at 0x00000000 in 0.4 seconds (82.3 kbit/s)...
Erasing flash...
Took 0.16s to erase flash block
Writing at 0x00001000... (50 %)
Writing at 0x00001400... (100 %)
Wrote 2048 bytes at 0x00001000 in 0.2 seconds (83.2 kbit/s)...
Erasing flash...
Took 2.33s to erase flash block
Writing at 0x00002000... (0 %)
Writing at 0x00002400... (0 %)
Writing at 0x00002800... (0 %)
Writing at 0x00002c00... (1 %)
Writing at 0x00003000... (1 %)
Writing at 0x00003400... (1 %)
Writing at 0x00003800... (2 %)
Writing at 0x00003c00... (2 %)
Writing at 0x00004000... (2 %)
Writing at 0x00004400... (3 %)
Writing at 0x00004800... (3 %)
Writing at 0x00004c00... (3 %)
Writing at 0x00005000... (4 %)
Writing at 0x00005400... (4 %)
Writing at 0x00005800... (4 %)
Writing at 0x00005c00... (5 %)
Writing at 0x00006000... (5 %)
Writing at 0x00006400... (5 %)
Writing at 0x00006800... (6 %)
Writing at 0x00006c00... (6 %)
Writing at 0x00007000... (6 %)
Writing at 0x00007400... (7 %)
Writing at 0x00007800... (7 %)
Writing at 0x00007c00... (7 %)
Writing at 0x00008000... (8 %)
Writing at 0x00008400... (8 %)
Writing at 0x00008800... (8 %)
Writing at 0x00008c00... (9 %)
Writing at 0x00009000... (9 %)
Writing at 0x00009400... (9 %)
Writing at 0x00009800... (10 %)
Writing at 0x00009c00... (10 %)
Writing at 0x0000a000... (10 %)
Writing at 0x0000a400... (11 %)
Writing at 0x0000a800... (11 %)
Writing at 0x0000ac00... (11 %)
Writing at 0x0000b000... (12 %)
Writing at 0x0000b400... (12 %)
Writing at 0x0000b800... (12 %)
Writing at 0x0000bc00... (13 %)
Writing at 0x0000c000... (13 %)
[... 401 lines cut ...]
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3fff5bd0, prio:14, stack:512
pm_task_hdl : 3fff5630, prio:1, stack:176
frc2_timer_task_hdl:0x3fffa1b0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jan 12 2018 03:24:49
phy ver: 273, pp ver: 8.3

Loading Punyforth
mode : softAP(ce:50:e3:7c:30:6e)
add if1
bcn 100
```
But the flash was not complete and a stack underflow (EUNDERFLOW) causes a restart (and flashing again with '--binary false' following #45 didn't change anything).

MayamaTakeshi, 2020-10-04:
I experimented replacing esptool.py with this one
  https://github.com/cefn/esptool/blob/master/esptool.py
mentioned in #40

It didn't work because after uploading the binaries it tries to hard reset the board before doing further steps.
But by ESP-01 requires soft_reset instead.

But since the flash.py doesn't expose the option "--after hard_reset|soft_reset|no_reset", I just modified hard_reset in esptool.py to do a soft_reset instead.
And with this, I could flash the board without errors:
```
takeshi@takeshi-brastel:bin$ python flash.py /dev/ttyUSB0 --flashmode dout
Selected modules: ALL
Flashing binaries..
Flashing 0x0: rboot.bin, 0x1000: blank_config.bin, 0x2000: punyforth.bin
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.4.0-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0340
Compressed 3104 bytes to 2169...
Wrote 3104 bytes (2169 compressed) at 0x00000000 in 0.2 seconds (effective 126.3 kbit/s)...
Hash of data verified.
Compressed 2048 bytes to 23...
Wrote 2048 bytes (23 compressed) at 0x00001000 in 0.0 seconds (effective 2172.9 kbit/s)...
Hash of data verified.
Compressed 310784 bytes to 203813...
Wrote 310784 bytes (203813 compressed) at 0x00002000 in 18.0 seconds (effective 138.4 kbit/s)...
Hash of data verified.

Leaving...
Soft resetting...
Flashing LAYOUT.tmp
esptool.py v2.4.0-dev
Connecting.....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x75fd77, SizeID=0x75), defaulting to 4MB
Compressed 392 bytes to 184...
Wrote 392 bytes (184 compressed) at 0x00051000 in 0.0 seconds (effective 141.0 kbit/s)...
Hash of data verified.

Leaving...
Soft resetting...
Flashing 0x52000: APP.tmp, 0x53000: CORE.tmp, 0x56000: DHT22.tmp, 0x57000: FLASH.tmp, 0x58000: FONT57.tmp, 0x59000: GPIO.tmp, 0x5a000: MAILBOX.tmp, 0x5b000: NETCON.tmp, 0x5d000: NTP.tmp, 0x5e000: PING.tmp, 0x5f000: SONOFF.tmp, 0x60000: SSD1306I2C.tmp, 0x62000: SSD1306SPI.tmp, 0x64000: TASKS.tmp, 0x65000: TCPREPL.tmp, 0x66000: TURNKEY.tmp, 0x67000: WIFI.tmp, 0x68000: EVENT.tmp, 0x69000: RINGBUF.tmp, 0x6a000: DECOMP.tmp, 0x6b000: PUNIT.tmp
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.4.0-dev
Connecting.....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 40 bytes to 39...
[... 121 lines cut ...]
Flash Size:   32 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3fff5bd0, prio:14, stack:512
pm_task_hdl : 3fff5630, prio:1, stack:176
frc2_timer_task_hdl:0x3fffa1b0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jan 12 2018 03:24:49
phy ver: 273, pp ver: 8.3

Loading Punyforth
mode : softAP(ce:50:e3:7c:30:6e)
add if1
bcn 100
.......
(stack)
```

I am assuming stack underflow should not restart the board since this is a REPL for experimentation.
Is this correct? 
Or is indeed a restart expected to happen? (well maybe it should happen in case of an app running as it could be exploited).

Jos-Ven, 2020-10-04:
> However, I still get a restart of the board if I force a stack underflow
That is normal under PunyForth.
PunyForth restarts as soon a there is a serious error.

see also: 
https://github.com/zeroflag/punyforth/issues/45
the comment of zeroflag on 18 Feb 2019

MayamaTakeshi, 2020-10-04:
Ah, so everything is fine.
Thanks.

closed 2020-10-04 by MayamaTakeshi