zeroflag/punyforth #60: Word Glossary I have written

bob-g4bby, 2020-10-21:
Attila,
I've written a glossary for punyforth in LibreOffice. I couldn't work out what a few of the words do, but it is about 99% done. You are welcome to include it with the code.
regards, Bob SW U.K.
[Glossary of Punyforth Words.docx](https://github.com/zeroflag/punyforth/files/5415750/Glossary.of.Punyforth.Words.docx)
[Glossary of Punyforth Words.pdf](https://github.com/zeroflag/punyforth/files/5415752/Glossary.of.Punyforth.Words.pdf)

zeroflag, 2020-10-21:
This is awesome, thanks for your work, I'm going to add it to to the repo.

bob-g4bby, 2020-10-22:
Hi Attila,
You're welcome, thank you for publishing Punyforth. It's been fun running (and crashing) small programs, to learn how stuff works. The glossary has all the words included but some descriptions are missing through lack of information. There is a small list of unsorted words you might like to place and descriptions as you like.
To talk to ESP8266 modules via UDP etc, I use the free community edition of Labview, a very powerful test and measurement ide I used for many years before I retired. Once you know it, like forth it saves a lot of time over conventional languages and runs fast. You get something running very quickly!

bob-g4bby, 2020-10-22:
[https://www.ni.com/en-gb/shop/labview/select-edition/labview-community-edition.html](url)

bob-g4bby, 2020-10-22:
Hi Attila,

You're welcome, thank you for publishing Punyforth. It's been fun running (and crashing) small programs, to learn how stuff works. The glossary has all the words included but some descriptions are missing through lack of information. There is a small list of unsorted words you might like to place and descriptions as you like.

To talk to ESP8266 modules via UDP etc, instead of Python I use the [LabView Community Edition](https://www.ni.com/en-gb/shop/labview/select-edition/labview-community-edition.html) a very powerful test and measurement language for Windows I used for many years before I retired. Once you know it, like forth it saves a lot of time over conventional languages and runs fast. You get something running very quickly!

zeroflag, 2020-10-25:
Thanks for your contribution, I included the glossaries in 32a3016.

closed 2020-10-25 by zeroflag