zeroflag/punyforth #64: Can TCP REPL eval be replicated using just strings?

kalizi, 2021-03-13:
Hi there, I'm using punyforth for a university project, an SPI Interface for a microSD card to read its content. 
After reading the content of a single block, I'm storing it to a 512B buffer and the teacher suggested extending the project by executing the code from that buffer. 
I read the eval definition on the tcp-repl and tried replicating it manually simple ASCII codes but it doesn't work, I probably didn't understand how it works and I'm totally doing it wrong. 

Can anyone explain me how should work and how can I do it?

I leave the code under here, 
Thanks in advance.

```forth
\ Code to execute:
\ 2 2 + 

\ ASCII:
\ 50 32 50 32 43

: dump-in 
	println: "Input stream: "
	cr
	tib 10 type-counted
	cr
;

: dump-tib
	print: "tib = " tib . cr 
;

: dump#tib 
	print: "#tib = " #tib @ . cr 
;

: dump>in 
	print: ">in = " >in . cr 
;

: test
	push-enter
	0 #tib ! 
	dump>in
	dump-tib
	tib >in !
	dump>in
	dump-tib

	50 chr>in dump#tib
	32 chr>in dump#tib
	50 chr>in dump#tib
	32 chr>in dump#tib
	43 chr>in dump#tib
	13 chr>in dump#tib
	10 chr>in dump#tib

	dump-in dump#tib dump>in

	push-enter

	dump-in dump#tib dump>in
	print: "Stack: " stack-print cr 
;

test
```

zeroflag, 2021-03-16:
Hi @kalizi

As far as I remember the repl functionality requires multi tasking mode.

```
multi
```

kalizi, 2021-03-16:
Thanks for your quick reply, I tried with this code:

```
\ Code to execute:
\ 2 2 + 

\ ASCII:
\ 50 32 50 32 43

0 task: main-task
0 task: exec-task

: test
	activate 
	begin
		0 #tib ! 
		tib >in !

		50 chr>in 
		32 chr>in 
		50 chr>in 
		32 chr>in 
		43 chr>in 
		13 chr>in 
		10 chr>in 

		push-enter

		print: "Stack: " stack-print cr 
		1000 ms
		pause
	again
	deactivate
;

: a 
	activate
	begin
		println: "Executing task 1..."
		1000 ms
		pause
	again
	deactivate
;

: multitask-start
	multi 
	main-task a 
	exec-task test 
;

multitask-start
```

And it worked. 

Thanks a lot!

closed 2021-03-16 by kalizi