zeroflag/punyforth #65: TCP-REPL triggers ESP8266 Restart?

Industry-Standard, 2021-05-07:
I tried on a NodeMCU Amica and an Adafruit Feather HUZZAH esp8266 but the behavior was the same. Calling repl-start throws an exception, and then the ESP reboots. _I am probably doing something wrong._ Any help would be greatly appreciated!

I followed the bouncing ball at https://github.com/zeroflag/punyforth and https://github.com/zeroflag/punyforth/wiki/Developing-and-deploying-Punyforth-applications, and did the following:

# Write some FORTH

Here's the contents of **wifi-repl.forth**, basically what I could tell as being enough to get a REPL over TCP. 

    EVENT       load
    TASKS       load
    MAILBOX     load
    FLASH       load
    TURNKEY     load
    NETCON      load
    WIFI        load
    TCPREPL     load

    println: "lookout world"
    "nwkey" "wifiNetwork" wifi-connect
    println: "connected?"
    repl-start

# Flash the ESP8266 

    box:~/code/punyforth/arch/esp8266/bin$ python flash.py /dev/ttyUSB0 --main ../forth/wifi-repl.forth

Flashes okay, at least seems to work; I get the UART REPL to load and can do some stuff. I'm flashing from Ubuntu MATE fwiw. 

# Connect to the UART REPL

    box:~$ picocom /dev/ttyUSB0 -b 115200 --omap crlf --echo --imap delbs

# Output when running
    ESP-Open-SDK ver: 0.0.1 compiled @ Jan 12 2018 03:24:49
    phy ver: 273, pp ver: 8.3
    
        
    Loading Punyforth
        mode : sta(a0:20:a6:13:18:0b)
    add if0
    ..scandone
    ..reconnect
    scandone
    ........add 0
    aid 3
    cnt 
    
    connected with wifiNetwork, channel 11
    dhcp client start...
    ...ip:192.168.1.121,mask:255.255.255.0,gw:192.168.1.1
    ................Fatal exception (2): 
    epc1=0x3ffeeb30
    epc2=0x00000000
    epc3=0x4023db1f
    excvaddr=0x3ffeeb30
    depc=0x00000000
    excsave1=0x40215a90
    Registers:
    a0 40215a90 a1 3fffbfd0 a2  0000000a a3  3fffa4d8
    a4  00064aa2 a5  00000aa2 a6  04000002 a7  3feffe00
    a8  3ffeeb54 a9  3ffeeb30 a10 ffffffff a11 3ffe88fc
    a12 00000000 a13 3ffeec48 SAR 0000001e
    
    Stack: SP=0x3fffbfd0
    0x3fffbfd0: 00000000 00000000 00000000 00000000
    0x3fffbfe0: 00000000 00000000 00000000 a5a5a5a5
    0x3fffbff0: eb48d176 0000014c 3fffbf80 000007c4
    
    Free Heap: 11592
    _heap_start 0x3fff7a68 brk 0x3fffd42c supervisor sp 0x40000000 sp-brk 11220 bytes
    arena (total_size) 22980 fordblks (free_size) 372 uordblocks (used_size) 22608
    
     ets Jan  8 2013,rst cause:2, boot mode:(3,6)
    
    load 0x40100000, len 2292, room 16 
    tail 4
    chksum 0x57
    load 0x3ffe8000, len 772, room 4 
    tail 0
    chksum 0x0b
    csum 0x0b
    
    rBoot v1.4.0 - richardaburton@gmail.com
    Flash Size:   32 Mbit
    Flash Mode:   QIO
    Flash Speed:  40 MHz
    rBoot Option: Big flash
    rBoot Option: RTC data
    
    Booting rom 0.
    Wcl�|�"
            ��o�p�܌No�lNn���p�rc2�o|
                                          Bp#$l pp_task_hdl : 3fff5c20, prio:14, stack:512
    pm_task_hdl : 3fff5680, prio:1, stack:176
    frc2_timer_task_hdl:0x3fff4950, prio:12, stack:200
    
    ESP-Open-SDK ver: 0.0.1 compiled @ Jan 12 2018 03:24:49
    phy ver: 273, pp ver: 8.3

Boot loop; it'll do this forever, whether it gets connected to the wifi AP or not.

zeroflag, 2021-05-07:
Can you try by not loading TASKS, WIFI, NETCON, MAILBOX? Because TCPREPL loads these directly or indirectly. Loading them manually multiple times will likely eat up the ram.

Industry-Standard, 2021-05-08:
Thanks for the quick reply on this. With the recommended changes, it works!
With the following code:

```
EVENT       load
FLASH       load
TURNKEY     load
TCPREPL     load

repl-start
```

I am able to connect to the TCP REPL and start playing around!

I wanted to ask about something in the guide: 

> The station mode Wi-Fi settings are persistently stored by the ESP8266, there is no need to setup the Wi-Fi at every startup.

As I understand it, you have to set up the Wi-Fi at least once for the station settings to persist. Is there a way to verify that the Wi-Fi is connected on subsequent boots of the ESP8266 to avoid having to execute the ```wifi-connect``` word on each boot?

closed 2021-05-08 by Industry-Standard

zeroflag, 2021-05-09:
Yes, you can set up the wifi once, and remove the wifi-connect from the code. If you try you make a connection without having wifi, it will throw an exception (ENETCON).