zeroflag/punyforth #69: Feature request: forth repl on tcp/ip or i2c

bangzek, 2023-03-29:
I want to use punyforth for getting some data from serial port but after browsing around there are some problem I found:
1. The way I understand it, effectively only 1 uart on esp8266 (the second uart is transmit only)
2. punyforth use that serial forth as repl
3. Is there anyway to move the repl on tcp/ip port? effectively making punyforth a telnet forth
4. or move the repl to i2c (which added benefit I can use several board on an i2c wire)
5. Another small but useful trick is adding softuart feature to punyforth but some of my device use some weird unusual baudrate 5787 8E1 so I worry if the softuart can't handle that use case

TIA

zeroflag, 2023-03-29:
Hi @bangzek there is a TCP REPL, you can read more about it in https://github.com/zeroflag/punyforth/wiki/Running-Punyforth-on-a-power-outlet.

bangzek, 2023-03-30:
Great!
Is there a way to make punyforth not using the first uart at all?

zeroflag, 2023-03-30:
> Is there a way to make punyforth not using the first uart at all?

I don't think so. AFAIK RTOS / ESP-IDF already uses the UART for printf(). If you restart your esp you'll see some messages which are not coming from punyforth and they're written out via UART.

There might be some configs to disable or customize it:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp-console-uart

But this might require recompilation of the binary.